arendst/Tasmota #4747: Problem when verifying code - esp-knx library

bjhutchins, 2018-12-27:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Development/Compiler/Upload tools used : Arduino Ide 1.8.5
- [x] Hardware used : D1 Mini
- [ ] If a pre-compiled release or development binary was used, which one? :
- [ ] You have tried latest release or development binaries? :
- [ ] Provide the output of command ``status 0`` :
```
/Users/BenHutchins/Documents/Arduino/libraries/esp-knx-ip-0.5.1/esp-knx-ip-send.cpp: In member function 'void ESPKNXIP::send_2byte_float(const address_t&, knx_command_type_t, float)':
/Users/BenHutchins/Documents/Arduino/libraries/esp-knx-ip-0.5.1/esp-knx-ip-send.cpp:139:22: error: invalid operands of types 'double' and 'int' to binary 'operator&'
  long m = round(v) & 0x7FF;
                      ^
exit status 1
Error compiling for board Generic ESP8266 Module.
```

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2018-12-27:
Sorry, Tasmota is compiling fine.

Please check your build environment

closed 2018-12-27 by ascillato2

bjhutchins, 2018-12-27:
Could you provide a bit more information on what might be causing the problem?

bjhutchins, 2018-12-27:
im using 2.5.0-beta1 board manager should i use a prior version?

ascillato, 2018-12-27:
Sorry, but this type of compiler issues were most of the time solved just uninstalling, deleting all related folders and installing again.

Tasmota is compiling fine under cores 2.3.0, 2.4.2 and 2.5.0

You can find the precompiled bins 

core 2.3.0
http://thehackbox.org/tasmota/release/020300/
core 2.4.2
http://thehackbox.org/tasmota/
core 2.50
http://thehackbox.org/tasmota/020500/

ascillato, 2018-12-27:
Tasmota is compiling fine also under core 2.5.0, please, check that you have latest Tasmota