arendst/Tasmota #4751: Unstable WiFi connection with sonoff devices using Tasmota 6.4.X version

mmatesic01, 2018-12-28:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

**Describe the bug**
Lately I was having a lot of ghost switching with my Sonoff devices (S20, TH16, R2POW). It was driving me nuts.
After a lot of checkups, I found out that my Sonoff devices were randomly and often losing WiFi connection with my router and that was that cause of my MQTT triggered ghost switches. And the WiFi signals are strong. 
All devices were having 6.4.X version of Tasmota. I tired every advice on the internet, with no positive results. 

My final step was to downgrade my devices to Tasmota 6.3.X version, cause I recollect that at the beginning I had no such issues. After that the problem was completely gone. Now everything is completely stable (WiFi connection) and I did not change anything in the configuration on any device, before or after the downgrade.

If someone else has a similar problem may this could be a solution. But the problem resides somewhere in the latest versions (6.4.X). At least in my case.

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x ] _Searched the problem in issues and in the wiki_
- [ ] _Hardware used_ : 
- [ ] _Development/Compiler/Upload tools used_ :
- [ ] _If a pre-compiled release or development binary was used, which one?_ :
- [x ] _You have tried latest release or development binaries?_ :
- [ ] _Provide the output of command_``status 0`` :
```
STATUS 0 OUTPUT HERE - DO NOT DELETE THE MARKERS ABOVE AND BELOW THIS LINE
```

**To Reproduce**
_Steps to reproduce the behavior:_

**Expected behavior**
_A clear and concise description of what you expected to happen._

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Additional context**
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

closed 2018-12-28 by mmatesic01

ascillato, 2018-12-28:
Hi,

Seems that you have a bad flashing problem.

You should erase all flash with esptool.py as explained in the wiki and flash again with the precompiled bins from release.

Please, check the wiki at https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#wifi-issues-arduino-core-versions-and-expressif-sdk for explanations
Also check issue: #4697

timdonovanuk, 2019-07-20:
@ascillato2 this link crops up everywhere on here but doesn't actually go to the intended article. Worse, there doesn't seem to be any wifi troubleshooting article that even exists anymore in the wiki.

meingraham, 2019-07-20:
The wiki was extensively restructured many months ago. You can use GitHub search to find information in the wiki. Two articles come to mind: Troubleshooting and FAQ.