arendst/Tasmota #4752: wifi disconnect and sleep command freeze device.

bennyente, 2018-12-28:
I'm on version 6.4.1 prebuild image sonoff.bin on Sonoff basic device upload via esptool.
The problem is wifi disconnect frequently and randomly.
I try to disable Sleep with command "sleep 0" from web console and the device freeze.
If I poweroff and poweron the device nothing happen.
I have restore factory by long press button for 40 sec.
Also just do "sleep" command from web console freeze the device and restore to default nedeed. 

13:45:12 WIF: Connect failed with AP incorrect password
13:45:13 WIF: Connecting to AP1 XXXXXXXXXX in mode 11N as lucesalotto...
13:45:24 WIF: Connected
13:45:24 DNS: Initialized
13:45:25 MQT: Attempting connection...
13:45:25 MQT: Connected
13:45:25 MQT: lucesalotto/tele/LWT = Online (retained)
13:45:25 MQT: lucesalotto/cmnd/POWER = 
13:46:02 DNS: Initialized
13:46:03 MQT: Attempting connection...
13:46:03 MQT: Connected
13:46:03 MQT: lucesalotto/tele/LWT = Online (retained)
13:46:03 MQT: lucesalotto/cmnd/POWER = 
13:47:13 DNS: Initialized
13:47:13 MQT: Attempting connection...
13:47:14 MQT: Connected
13:47:14 MQT: lucesalotto/tele/LWT = Online (retained)
13:47:14 MQT: lucesalotto/cmnd/POWER = 
13:47:19 MQT: lucesalotto/tele/STATE = {"Time":"2018-12-28T13:47:19","Uptime":"1T05:07:03","Vcc":3.499,"SleepMode":"Normal","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"XXXXXXXXXX","BSSId":"C0:56:27:B5:D7:B7","Channel":6,"RSSI":54}}
13:47:52 MQT: Attempting connection...
13:47:52 MQT: Connect failed to 10.xx.xx.xx:1883, rc -2. Retry in 10 sec
13:47:55 DNS: Initialized
13:48:06 MQT: Attempting connection...
13:48:06 MQT: Connected
13:48:06 MQT: lucesalotto/tele/LWT = Online (retained)
13:48:06 MQT: lucesalotto/cmnd/POWER = 
13:50:10 MQT: Attempting connection...
13:50:10 MQT: Connect failed to 10.xx.xx.xx:1883, rc -2. Retry in 10 sec
13:50:24 WIF: Connect failed with AP incorrect password
13:50:24 WIF: Connecting to AP2 XXXXXXXXXX in mode 11N as lucesalotto...
13:50:35 WIF: Connected
13:50:35 DNS: Initialized
13:50:46 MQT: Attempting connection...
13:50:46 MQT: Connected
13:50:46 MQT: lucesalotto/tele/LWT = Online (retained)
13:50:46 MQT: lucesalotto/cmnd/POWER = 
13:51:35 WIF: Connect failed with AP incorrect password
13:51:35 WIF: Connecting to AP1 XXXXXXXXXX in mode 11N as lucesalotto...
13:51:39 WIF: Connected
13:51:39 DNS: Initialized
13:51:40 MQT: Attempting connection...
13:51:40 MQT: Connected
13:51:40 MQT: lucesalotto/tele/LWT = Online (retained)
13:51:40 MQT: lucesalotto/cmnd/POWER = 
13:52:19 MQT: lucesalotto/tele/STATE = {"Time":"2018-12-28T13:52:19","Uptime":"1T05:12:03","Vcc":3.498,"SleepMode":"Normal","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"XXXXXXXXXX","BSSId":"A4:91:B1:22:7B:1D","Channel":1,"RSSI":54}}
 
State 0 command:

13:57:06 CMD: status 0
13:57:06 MQT: lucesalotto/stat/STATUS = {"Status":{"Module":1,"FriendlyName":["lucesalotto"],"Topic":"lucesalotto","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
13:57:06 MQT: lucesalotto/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-IT.bin","RestartReason":"Software/System restart","Uptime":"1T05:16:50","StartupUTC":"2018-12-27T07:40:16","Sleep":50,"BootCount":11,"SaveCount":351,"SaveAddress":"F8000"}}
13:57:06 MQT: lucesalotto/stat/STATUS2 = {"StatusFWR":{"Version":"6.4.1(sonoff)","BuildDateTime":"2018-12-24T14:41:00","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
13:57:06 MQT: lucesalotto/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["XXXXXXXXXX","XXXXXXXXXX"],"TelePeriod":300,"SetOption":["000A8009","558180C0","00000620"]}}
13:57:06 MQT: lucesalotto/stat/STATUS4 = {"StatusMEM":{"ProgramSize":521,"Free":480,"Heap":17,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
13:57:06 MQT: lucesalotto/stat/STATUS5 = {"StatusNET":{"Hostname":"lucesalotto","IPAddress":"10.12.77.142","Gateway":"10.12.77.1","Subnetmask":"255.255.255.0","DNSServer":"10.12.77.2","Mac":"5C:CF:7F:71:C3:3C","Webserver":2,"WifiConfig":4}}
13:57:06 MQT: lucesalotto/stat/STATUS6 = {"StatusMQT":{"MqttHost":"10.xx.xx.xx","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_71C33C","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
13:57:06 MQT: lucesalotto/stat/STATUS7 = {"StatusTIM":{"UTC":"Fri Dec 28 12:57:06 2018","Local":"Fri Dec 28 13:57:06 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":"+01:00","Sunrise":"08:43","Sunset":"17:00"}}
13:57:06 MQT: lucesalotto/stat/STATUS10 = {"StatusSNS":{"Time":"2018-12-28T13:57:06"}}
13:57:06 MQT: lucesalotto/stat/STATUS11 = {"StatusSTS":{"Time":"2018-12-28T13:57:06","Uptime":"1T05:16:50","Vcc":3.500,"SleepMode":"Normal","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"XXXXXXXXXX","BSSId":"A4:91:B1:22:7B:1D","Channel":1,"RSSI":50}}}

Jason2866, 2018-12-28:
Do you have erased flash completly with esptool before flashing Tasmota?
It looks like your issues comes from fragments left in flash.

ascillato2, 2018-12-28:
Hi,

Seems that you have a bad flashing problem.

You should erase all flash with esptool.py as explained in the wiki and flash again with the precompiled bins from release.

Please, check the wiki at https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#wifi-issues-arduino-core-versions-and-expressif-sdk for explanations
Also check issue: #4697

closed 2018-12-28 by ascillato2

bennyente, 2018-12-28:
Hi, thanks, but I don’t think so.
I flash again with esptool one of my Sonoff and have the same issue too.

bennyente, 2018-12-28:
Hi, thanks, but I don’t think so.
I flash again with esptool one of my Sonoff and have the same issue too.

Benedetto Napoletano
benedetto.napoletano@gmail.com

Jason2866, 2018-12-28:
Have you erased the whole flash BEFORE flashing Tasmota?
Just flashing Tasmota doesnt help!

bennyente, 2018-12-28:
Yes,

I do:
1. esptool.py --port COM9 erase_region 0x0F4000 0x008000
2. esptool.py --port COM9 erase_flash
3. esptool.py --port COM9 write_flash -fs 1MB -fm dout 0x0 sonoff.bin

Jason2866, 2018-12-28:
Try http://thehackbox.org/tasmota/020500/sonoff.bin

ascillato, 2018-12-28:
The erase region will not work. You need to erase full flash. You are leaving without erasing the sdk memory region where your issue remain.

ascillato, 2018-12-28:
Please, read the link provided

bennyente, 2018-12-29:
with 2.5.0 core I have the same problem. Now I have installed 6.4.1/2.3.0 from here http://thehackbox.org/tasmota/release/020300/

Jason2866, 2018-12-29:
Do you have a repeater or a Mesh setup?
Do you use Ubiquity hardware?

bennyente, 2018-12-29:
i see the core 2.4.2 and 2.5.0 set Vcc to 3,5V but the core 2.3.0 set Vcc to 3.17V. Maybe the over volt cause chipset instability.

bennyente, 2018-12-29:
> Do you have a repeater or a Mesh setup?
> Do you use Ubiquity hardware?

i have a repeater. but i also try to blacklist the Sonoff MAC address on repeater so the device connect to main access point. but the problem remain.

Jason2866, 2018-12-29:
The cores are calculating the measured voltage on analog input different.
This is normal...

bennyente, 2018-12-31:
Hi, after 2 days test, I can confirm with firmware 6.4.1/2.3.0 all working fine now.