arendst/Tasmota #4756: Still Bug with WIFI (see #4721)

ymurawski, 2018-12-29:
Hi, i still have the Problem with disconnects like in my post 

https://github.com/arendst/Sonoff-Tasmota/issues/4721

I deleted the flash but i doesnt helped for me.
When i deactivate the Sonoff adapter in Iobroker everything works.
Maybe it is a bug when emulation Hue Bridge ?

Thanks alot.

**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

**Describe the bug**
_A clear and concise description of what the bug is._

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [ ] _Searched the problem in issues and in the wiki_
- [ ] _Hardware used_ : 
- [ ] _Development/Compiler/Upload tools used_ :
- [ ] _If a pre-compiled release or development binary was used, which one?_ :
- [ ] _You have tried latest release or development binaries?_ :
- [ ] _Provide the output of command_``status 0`` :
```
STATUS 0 OUTPUT HERE - DO NOT DELETE THE MARKERS ABOVE AND BELOW THIS LINE
```

**To Reproduce**
_Steps to reproduce the behavior:_

**Expected behavior**
_A clear and concise description of what you expected to happen._

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Additional context**
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2018-12-29:
Sorry i cant reproduce your issue. 
A place to search is your iobroker, the issues are starting when you activate things here....

ymurawski, 2018-12-30:
please reactivate my old post :)