arendst/Tasmota #4758: Power Toggle

serapis80, 2018-12-29:
Hello to all,
I'm writing because I'm looking for a function that I want to use in Sonoff 4ch pro device in order to simulate a diverter switch.

I already installed the last version of Tasmota firmware 6.4.1 prebuild and I tried to check if it's possible to toggle the relay with Alexa, but seems that it not present like a function.

I want to integrate the sonoff 4ch pro device in the electrical system of my home like the picture below.

[image]

One relay of the sonoff 4ch pro can be used like a diverter and the other one it's the common diverter used by finger. In this way I can use my voice from Alexa and my finger (in case of wifi or internet is not available) in order to switch on and off the light.

The problem can appears when I "switch on" the light from the hardware diverter and I want to "switch off" the light from Alexa. In this case I say to Alexa: "turn on the light" in order to switch off.

So, I want to say to Alexa "turn on the light" and inside the Sonoff 4ch pro, the code read the actual status of the relay and change this state (if the actual status of the relay is to 0, it will change to 1 and viceversa).

For Alexa I use the Hue Emulator, and I tried to modify the code in xplg_wemohue.ino like this:
```
switch(on)
        {
          case false : ExecuteCommandPower(device, POWER_TOGGLE, SRC_HUE); // POWER_OFF
                       response.replace("{re", "false");
                       break;
          case true  : ExecuteCommandPower(device, POWER_TOGGLE, SRC_HUE);     // POWER_ON
                       response.replace("{re", "true");
                       break;
          default    : response.replace("{re", (power & (1 << (device-1))) ? "true" : "false");
                       break;
        }
```
Just a simple POWER_TOGGLE instead POWER_OFF or POWER_ON.
Compiled and download to the device, the relay works fine in all condictions, but in some cases the reply to Alexa is not in according to the request. The relay change the state but Alexa reply with "Seems that the device not works properly" (perhaps the sentence in English is not the same, sorry but I use a different language).
I try to explain better the condiction here below:

[image]

So, the relay works fine, but my gol is complete at 99%. For complete at 100% I need your help.
I not understand how the Tasmota Firmware can reply with a different value.
In the code posted above, the response is fixed to False or True, but seems that the firmware read the actual status of the relay and reply with this value, but where ?

If I write in a browser the string:
`http://192.168.XXX.XXX/api/username/lights/1/state?1={"on":true}`
the device reply to me with : 
`[{"success":{"/lights/1/state/on":true}}]`
The relay change the status every time that I send the string, and the reply will be always : 
`[{"success":{"/lights/1/state/on":true}}]`

The same if I use the string:
`http://192.168.137.55/api/username/lights/1/state?1={"on":false}`
the reply will be always:
`[{"success":{"/lights/1/state/on":false}}]`

Someone can tell me the part of the code can reply to Alexa in order to change an reply in according to the request ?

Thank you in advance for your support.
Best Regards

serapis80, 2018-12-29:
I forgot this :

- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Development/Compiler/Upload tools used : Arduino 1.8.8
- [X] Hardware used : Sonoff 4ch pro
- [X] If a pre-compiled release or development binary was used, which one? : development  release based on 6.4.1 version
- [X] You have tried latest release or development binaries? : tried the last one
- [X] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
16:38:22 CMD: status 0
16:38:22 RSL: stat/sonoff/STATUS = {"Status":{"Module":23,"FriendlyName":["Sonoff","Sonoff2","Sonoff3","Sonoff4"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
16:38:22 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Power on","Uptime":"0T22:56:26","StartupUTC":"2018-12-28T16:41:56","Sleep":50,"BootCount":2,"SaveCount":75,"SaveAddress":"F9000"}}
16:38:22 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.4.1(sonoff)","BuildDateTime":"2018-12-28T17:37:33","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
16:38:22 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Kowalski",""],"TelePeriod":300,"SetOption":["00008009","55C180C0","00000000"]}}
16:38:22 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":524,"Free":476,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
16:38:22 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-7611","IPAddress":"192.168.137.55","Gateway":"192.168.137.254","Subnetmask":"255.255.255.0","DNSServer":"10.0.0.138","Mac":"84:0D:8E:46:1D:BB","Webserver":2,"WifiConfig":4}}
16:38:22 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_461DBB","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
16:38:22 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Sat Dec 29 15:38:22 2018","Local":"Sat Dec 29 16:38:22 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":"+01:00","Sunrise":"08:43","Sunset":"17:00"}}
16:38:22 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-12-29T16:38:22"}}
16:38:22 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-12-29T16:38:22","Uptime":"0T22:56:26","Vcc":3.434,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":114,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"Kowalski","BSSId":"60:E3:27:F0:18:AD","Channel":1,"RSSI":66}}}
```

Regards

serapis80, 2018-12-29:
Hello,
it's possible that Alexa requests again the status with HueLightStatus1 ?
```
void HueLightStatus1(byte device, String *response)
{
  float hue = 0;
  float sat = 0;
  float bri = 254;
  uint16_t ct = 500;

  if (light_type) {
    LightGetHsb(&hue, &sat, &bri, g_gotct);
    ct = LightGetColorTemp();
  }
  *response += FPSTR(HUE_LIGHTS_STATUS_JSON);
  response->replace("{state}", (power & (1 << (device-1))) ? "true" : "false");
  response->replace("{h}", String((uint16_t)(65535.0f * hue)));
  response->replace("{s}", String((uint8_t)(254.0f * sat)));
  response->replace("{b}", String((uint8_t)(254.0f * bri)));
  response->replace("{t}", String(ct));
  response->replace("{m}", g_gotct?"ct":"hs");
}
```

I will try to modify the code in order to reply the value in accordance with the last command request when Alexa request "HueLightStatus1".

I'll keep you posted

Regards

ascillato2, 2018-12-29:
Hi, There is no need to modify anything in the code.

what you want to achieve is called interlock and it is already implemented in Tasmota

if you go to the console and type `SetOption14 1` your device will behave as you asked, only one relay to be on all the time. See for more information: https://github.com/arendst/Sonoff-Tasmota/wiki/Commands

closed 2018-12-29 by ascillato2