arendst/Tasmota #4759: DS3231 shown as HX711

mvincm, 2018-12-29:
Hello!

**Describe the bug**
I have compiled tasmota (6.4.1.2) for "sonoff basic" with DS3231 option enabled in "my_user_config.h"(uncomented line #define USE_DS3231). I use GIPO5 (SCL) and GPIO4 (SDA) with a little change in "sonoff_template.h": 
```
GPIO_USER,        // GPIO04 Optional sensor 
GPIO_USER,        // GPIO05 Optional sensor
```
And RTC seems to work but there is some strange log about HX711 on a console and on the main page there is a label "HX711 Weight 0 kg"
```
00:00:00 Project sonoff Sonoff Version 6.4.1.2(sonoff)-2_4_2
00:00:00 I2C: DS3231 found at 0x68
00:00:00 WIF: Connecting to AP1 KoroFlat in mode 11N as sonoff-6333...
00:00:01 Set time from DS3231 to RTC (UTC) Sat Dec 29 16:08:41 2018, (DST) Sun Mar 25 02:00:00 2018, (STD) Sun Oct 28 03:00:00 2018
17:08:48 Write Time TO DS3231 from NTP (UTC) Sat Dec 29 16:08:48 2018, (DST) Sun Mar 25 02:00:00 2018, (STD) Sun Oct 28 03:00:00 2018
17:08:49 WIF: Connected
17:08:49 DNS: Initialized
17:08:49 HTP: Web server active on sonoff-6333.local with IP address 192.168.22.170
17:08:49 RSL: INFO1 = {"Module":"Sonoff Basic","Version":"6.4.1.2(sonoff)","FallbackTopic":"cmnd/DVES_9398BD_fb/","GroupTopic":"sonoffs"}
17:08:49 RSL: INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-6333","IPAddress":"192.168.22.170"}
17:08:49 RSL: INFO3 = {"RestartReason":"Software/System restart"}
17:08:49 RSL: RESULT = {"POWER":"OFF"}
17:08:49 RSL: POWER = OFF
17:09:00 RSL: STATE = {"Time":"2018-12-29T17:09:00","Uptime":"0T00:00:17","Vcc":3.242,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":275,"POWER":"OFF","Wifi":{"AP":1,"SSId":"KoroFlat","BSSId":"34:CE:00:63:20:41","Channel":6,"RSSI":100}}
17:09:00 RSL: SENSOR = {"Time":"2018-12-29T17:09:00","HX711":{"Weight":0}}
17:14:00 RSL: STATE = {"Time":"2018-12-29T17:14:00","Uptime":"0T00:05:17","Vcc":3.240,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":267,"POWER":"OFF","Wifi":{"AP":1,"SSId":"KoroFlat","BSSId":"34:CE:00:63:20:41","Channel":6,"RSSI":76}}
17:14:00 RSL: SENSOR = {"Time":"2018-12-29T17:14:00","HX711":{"Weight":0}}
17:18:43 APP: Serial logging disabled
17:19:00 RSL: STATE = {"Time":"2018-12-29T17:19:00","Uptime":"0T00:10:17","Vcc":3.226,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":274,"POWER":"OFF","Wifi":{"AP":1,"SSId":"KoroFlat","BSSId":"34:CE:00:63:20:41","Channel":6,"RSSI":78}}
17:19:00 RSL: SENSOR = {"Time":"2018-12-29T17:19:00","HX711":{"Weight":0}}
```

Something is wrong but kill me I do not know what... Compilation goes ok, RTC seems to work..

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x] _Searched the problem in issues and in the wiki_
- [x] _Hardware used_ : Sonoff basic R2 V1 
- [x] _Development/Compiler/Upload tools used_ : esptool
- [x] _If a pre-compiled release or development binary was used, which one?_ : development version 6.4.1.2
- [x] _You have tried latest release or development binaries?_ : yes
- [ ] _Provide the output of command_``status 0`` :
```
17:34:00 CMD: status 0
17:34:00 RSL: STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
17:34:00 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:25:17","StartupUTC":"2018-12-29T16:08:43","Sleep":50,"BootCount":12,"SaveCount":39,"SaveAddress":"F5000"}}
17:34:00 RSL: STATUS2 = {"StatusFWR":{"Version":"6.4.1.2(sonoff)","BuildDateTime":"2018-12-29T16:26:23","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
17:34:00 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["KoroFlat",""],"TelePeriod":300,"SetOption":["00008001","558180C0","00000000"]}}
17:34:00 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":522,"Free":480,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"146085","FlashMode":3,"Features":["00000809","0FDAE794","240383A0","23B617CE","00003FC0"]}}
17:34:00 RSL: STATUS5 = {"StatusNET":{"Hostname":"sonoff-6333","IPAddress":"192.168.22.170","Gateway":"192.168.22.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.22.1","Mac":"84:F3:EB:93:98:BD","Webserver":2,"WifiConfig":4}}
17:34:00 RSL: STATUS7 = {"StatusTIM":{"UTC":"Sat Dec 29 16:34:00 2018","Local":"Sat Dec 29 17:34:00 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":"+01:00","Sunrise":"08:43","Sunset":"17:00"}}
17:34:00 RSL: STATUS10 = {"StatusSNS":{"Time":"2018-12-29T17:34:00","HX711":{"Weight":0}}}
17:34:00 RSL: STATUS11 = {"StatusSTS":{"Time":"2018-12-29T17:34:00","Uptime":"0T00:25:17","Vcc":3.162,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":259,"POWER":"OFF","Wifi":{"AP":1,"SSId":"KoroFlat","BSSId":"34:CE:00:63:20:41","Channel":6,"RSSI":92}}}
```

**To Reproduce**
As described beyond

**Expected behavior**
No information about HX711 and maybe RTC time displayed on the main page.

Best regards,
MvincM

Jason2866, 2018-12-29:
Your change in `sonoff_template.h` is probably wrong.

mvincm, 2018-12-29:
It was one simple change. Nothing to do with HX711. Take a look:

Original file:
```
// Default module settings
const mytmplt kModules[MAXMODULE] PROGMEM = {
  { "Sonoff Basic",    // Sonoff Basic (ESP8266)
     GPIO_KEY1,        // GPIO00 Button
     GPIO_USER,        // GPIO01 Serial RXD and Optional sensor
     GPIO_USER,        // GPIO02 Only available on newer Sonoff Basic R2 V1
     GPIO_USER,        // GPIO03 Serial TXD and Optional sensor
     GPIO_USER,        // GPIO04 Optional sensor
     0,                // GPIO05
     0,                // GPIO06 (SD_CLK   Flash)
     0,                // GPIO07 (SD_DATA0 Flash QIO/DIO/DOUT)
     0,                // GPIO08 (SD_DATA1 Flash QIO/DIO/DOUT)
     0,                // GPIO09 (SD_DATA2 Flash QIO)
     0,                // GPIO10 (SD_DATA3 Flash QIO)
     0,                // GPIO11 (SD_CMD   Flash)
     GPIO_REL1,        // GPIO12 Red Led and Relay (0 = Off, 1 = On)
     GPIO_LED1_INV,    // GPIO13 Green Led (0 = On, 1 = Off)
     GPIO_USER,        // GPIO14 Optional sensor
     0,                // GPIO15
     0,                // GPIO16
     0                 // ADC0 Analog input
  },
```

Changed file:
```
const mytmplt kModules[MAXMODULE] PROGMEM = {
  { "Sonoff Basic",    // Sonoff Basic (ESP8266)
     GPIO_KEY1,        // GPIO00 Button
     GPIO_USER,        // GPIO01 Serial RXD and Optional sensor
     GPIO_USER,        // GPIO02 Only available on newer Sonoff Basic R2 V1
     GPIO_USER,        // GPIO03 Serial TXD and Optional sensor
     GPIO_USER,        // GPIO04 Optional sensor
     GPIO_USER,        // GPIO05 Optional sensor
     0,                // GPIO06 (SD_CLK   Flash)
     0,                // GPIO07 (SD_DATA0 Flash QIO/DIO/DOUT)
     0,                // GPIO08 (SD_DATA1 Flash QIO/DIO/DOUT)
     0,                // GPIO09 (SD_DATA2 Flash QIO)
     0,                // GPIO10 (SD_DATA3 Flash QIO)
     0,                // GPIO11 (SD_CMD   Flash)
     GPIO_REL1,        // GPIO12 Red Led and Relay (0 = Off, 1 = On)
     GPIO_LED1_INV,    // GPIO13 Green Led (0 = On, 1 = Off)
     GPIO_USER,        // GPIO14 Optional sensor
     0,                // GPIO15
     0,                // GPIO16
     0                 // ADC0 Analog input
  },
```

ascillato2, 2018-12-30:
Hi,

If you are not using the HX711 sensor, please, just disable it in _my_user_config.h_ file.

Thanks

ascillato2, 2018-12-30:
Have you defined the HX711 pins in the GPIO configuration ?

[image]

If I don't define those pins I don't have any readings from HX711.

mvincm, 2018-12-30:
Hello,

About "#define USE_HX711 like //#define USE_HX711". I will try it today and it should works (theoretically) but I'm afraid that there is some bug which could drain CPU/power (by not necessary readings/events).

I don't set up any HX171 sensors (thru web gui or thru my_user_config.h/sonoff_template.h file). The second observation is when I disconnect DS3231 those strange HX711 readings despair. So maybe it is problem with wrong I2C setup... but DS3231 works just fine.

Jason2866, 2018-12-30:
@ascillato2 
I can confirm it is a bug. If HX711 support is enabled there is HX711 on main page displayed....
[image]
[image]

arendst, 2018-12-30:
I will investigate

arendst, 2018-12-30:
Caused by DS3231 breaking chain of functions.

Repaired and should be fine now.

Jason2866, 2018-12-30:
@arendst  Thank you, can confirm works correct now.
@mvincm Thx for finding a bug, please close the issue.

mvincm, 2018-12-30:
Yeep!! It works just fine! Thx for quick response and support!

closed 2018-12-30 by mvincm

chedxb, 2019-06-23:
I have a similar problem, DS3231 is shown as BH1750, SHT3X-0x45 Temperature and SHT3X-0x45 Temperature.

I am trying to connect DS3231 to Sonoff Basic via serial as follows;

GPIO1 Serial Out        SDA
GPIO3 Serial In           SCL

Serial logging disabled. 

log during boot up is as follows;

00:00:00 Project sonoff Konteynir Bahce Isiklari Version 6.5.0(sonoff)-2_3_0
00:00:00 I2C: DS3231 found at 0x68
00:00:00 WIF: Connecting to AP1 Sur in mode 11N as sonoff-8166...
00:00:00 Set time from DS3231 to RTC (UTC) Wed Dec 31 00:00:00 1969, (DST) Sun Feb 15 19:31:44 1970, (STD) Sun Sep 20 20:31:44 1970
09:28:23 WIF: Connected
09:28:23 HTP: Web server active on sonoff-8166 with IP address 192.168.74.36
19:14:22 Write Time TO DS3231 from NTP (UTC) Sun Jun 23 16:14:22 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
19:15:19 CMD: I2Cscan
19:15:19 RSL: stat/sonoff/RESULT = {"I2CScan":"Error 4 at 0x02"}

Jason2866, 2019-06-23:
Gpio1 and 3 are not possible for a lot of use cases. See https://github.com/arendst/Sonoff-Tasmota/wiki/Expanding-Sonoffs#usable-pins

chedxb, 2019-06-23:
is there a way to use DS3231 with Sonoff Basic? 

As far as I know, gpio1 and 3 are the only option available on board, other than soldering directly to ESP8x chip.

Jason2866, 2019-06-23:
As you wrote correct only way is to solder on Esp for 1 additional Gpio
There is GPIO14 or Gpio2 (depends on rev.) Wired to the header.

chedxb, 2019-06-23:
there is GPIO14 wired to the header, what are the pin configuration in this case?

chedxb, 2019-06-24:
it worked on GPIO1 Serial Out SDA and GPIO3 Serial In SCL after adding forward slashes to other I2C devices (basically I kept only #define USE_DS3231), built and uploaded. Serial logging disabled.