arendst/Tasmota #4761: Random dropouts with V6.4.1, V6.30 OK

sheppy99, 2018-12-29:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

**Describe the bug**
_A clear and concise description of what the bug is._
Repeated random LWT offline messages for a few seconds.

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x] _Searched the problem in issues and in the wiki_
- [x] _Hardware used_ : SOnOff PoW V1 and SOnOffS20
- [x] _Development/Compiler/Upload tools used_ : Prebuilt Binary Downloaded from Wiki
- [x] _If a pre-compiled release or development binary was used, which one?_ : sonoff.bin
- [x] _You have tried latest release or development binaries?_ : no, reverted to V6.3.0
- [x] _Provide the output of command_``status 0`` :
```
STATUS 0 OUTPUT HERE - DO NOT DELETE THE MARKERS ABOVE AND BELOW THIS LINE
```
Taken with firmware reverted to V.6.3.0
20:42:18 CMD: Status 0
20:42:18 MQT: stat/BugKiller/STATUS = {"Status":{"Module":8,"FriendlyName":["BugKiller"],"Topic":"BugKiller","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
20:42:18 MQT: stat/BugKiller/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:29:18","StartupUTC":"2018-12-29T19:13:00","Sleep":50,"BootCount":36,"SaveCount":260,"SaveAddress":"F6000"}}
20:42:18 MQT: stat/BugKiller/STATUS2 = {"StatusFWR":{"Version":"6.3.0","BuildDateTime":"2018-10-30T17:33:53","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
20:42:18 MQT: stat/BugKiller/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SR3",""],"TelePeriod":300,"SetOption":["00008009","558180C0","00000000"]}}
20:42:18 MQT: stat/BugKiller/STATUS4 = {"StatusMEM":{"ProgramSize":489,"Free":512,"Heap":14,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","000183A0","23B617CE","00003BC0"]}}
20:42:18 MQT: stat/BugKiller/STATUS5 = {"StatusNET":{"Hostname":"BugKiller","IPAddress":"192.168.1.93","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"68:C6:3A:94:E5:F0","Webserver":2,"WifiConfig":5}}
20:42:18 MQT: stat/BugKiller/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.69","MqttPort":1886,"MqttClientMask":"BugKiller","MqttClient":"BugKiller","MqttUser":"BugKiller","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
20:42:18 MQT: stat/BugKiller/STATUS7 = {"StatusTIM":{"UTC":"Sat Dec 29 19:42:18 2018","Local":"Sat Dec 29 20:42:18 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"08:43","Sunset":"17:00"}}
20:42:18 MQT: stat/BugKiller/STATUS10 = {"StatusSNS":{"Time":"2018-12-29T20:42:18"}}
20:42:18 MQT: stat/BugKiller/STATUS11 = {"StatusSTS":{"Time":"2018-12-29T20:42:18","Uptime":"0T00:29:18","Vcc":3.206,"POWER":"OFF","Wifi":{"AP":1,"SSId":"SR3","BSSId":"B4:FB:E4:74:7F:39","Channel":11,"RSSI":100}}}
**To Reproduce**
_Steps to reproduce the behavior:_
Upgrade many devices to V6.4.1, monitor LWT messages and observe random dropouts

**Expected behavior**
_A clear and concise description of what you expected to happen._
No random dropouts in normal use

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Additional context**
_Add any other context about the problem here._
Very sort range WiFi, so range problems unlikely, brand new Access Point located nearby, same problem with previous WiFi Access point which I just replaced. 100% signal showing on info screen

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2018-12-29:
Duplicate Issue see https://github.com/arendst/Sonoff-Tasmota/issues/4697

sheppy99, 2018-12-29:
Odd how the other issue didn’t come up when I searched. Will Monitor the other thread to see if a fix emerges

sheppy99, 2018-12-29:
Although that issue is closed so maybe that explains it.
Please advise which issue is the active one to follow