arendst/Tasmota #4762: RF Receiver not working/recognized

betyar, 2018-12-30:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Development/Compiler/Upload tools used :
- [X] Hardware used : ESP-01
- [X] If a pre-compiled release or development binary was used, which one? : precompiled
- [X] You have tried latest release or development binaries? :
- [X] Provide the output of command ``status 0`` :
01:23:35 MQT: stat/sonoff/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
01:23:35 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:04:48","StartupUTC":"2018-12-30T00:18:47","Sleep":50,"BootCount":16,"SaveCount":38,"SaveAddress":"F4000"}}
01:23:35 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.4.1(sonoff)","BuildDateTime":"2018-12-24T14:41:00","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
01:23:35 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["betyar0",""],"TelePeriod":300,"SetOption":["00008009","558180C0","00000001"]}}
01:23:35 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":521,"Free":480,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440E0","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
01:23:35 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-1535","IPAddress":"192.168.1.123","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"8.8.8.8","Mac":"A0:20:A6:23:E5:FF","Webserver":2,"WifiConfig":3}}
01:23:35 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.104","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_23E5FF","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
01:23:35 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Sun Dec 30 00:23:35 2018","Local":"Sun Dec 30 01:23:35 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":"+01:00","Sunrise":"08:43","Sunset":"17:01"}}
01:23:35 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-12-30T01:23:35"}}
01:23:35 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-12-30T01:23:35","Uptime":"0T00:04:48","Vcc":3.292,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"betyar0","BSSId":"00:23:CD:11:B1:9C","Channel":6,"RSSI":90}}}
**(Please, remember to close the issue when the problem has been addressed)**

I am building a simple Wifi/RF relay switch (similar to a Sonoff Basic RF) using an ESP-01 module. The GPIO connections are as follows (Tasmota settings in brackets):

GPIO0 - button (Button1 - 17)
GPIO2 - RF receiver (RFrecv - 106)
GPIO3 - relay (Relay1 - 21); the relay was taken from a bricked Sonoff Basic

I have a RX480-E4 4 channel RF Remote Control Receiver Module (1527 Encoding ) like this one: https://www.aliexpress.com/item/Hot-433mhz-Wireless-4-Channel-RF-Receiver-1527-Learning-Code-Decoder-Module-For-Remote-Control/32888969634.html?spm=a2g0s.9042311.0.0.27424c4dHwpjjS. 

Everything works except for the RF receiver. I have the latest firmware (6.4.1 sonoff) and have it configured for Generic (18). The relay can be switched on and off via the web interface and the button. Yet no matter what I tried the RF input doesn't seem to be recognized. The console doesn't show anything. I know the module is working because I had no problems pairing with the remote and when I press a button on the remote the LED on the board flashes. I also attached a LED to the VT pin and that also lights up when the the remote is pressed.

I tried different GPIOs but it's the same. I also tried different configurations, for example having it figured as a switch or a button (I read somewhere that the output of this module is like a simple switch so I thought that might work) but no luck. I also tried using the different data pins on the receiver but the result is still the same: nothing

Any suggestions?

Jason2866, 2018-12-30:
You have a wrong receiver this one is not compatible.
This one just use fixed! learned codes. 
As example this one is compatible
https://www.aliexpress.com/item/Transmitter-Module-Relay-Receiver-315MHz-Code-Learning-433MHz-Remote-Control-Transmitter-Receiver-Module/32825621368.html?ws_ab_test=searchweb0_0,searchweb201602_4_10320_10065_10068_10547_319_317_10548_10696_453_10084_454_10083_10618_10304_10307_10820_10821_538_537_10302_536_10843_10059_10884_10887_100031_10319_321_322_10103,searchweb201603_51,ppcSwitch_0&algo_expid=ddefd5a9-4bdf-4e59-ac33-5f6b3738eda4-2&algo_pvid=ddefd5a9-4bdf-4e59-ac33-5f6b3738eda4&transAbTest=ae803_3

betyar, 2018-12-30:
Thanks, I didn't realize this. I am new to this so there's something  I don't understand. The module I have has a small button through which I can pair my remote (that is why I bought it). The module you referred to doesn't have this, so how would I pair it with my remote? On a Sonoff RF this is done through the button on GPIO0. Since I have a button on GPIO0 and it works as it should (it switches the relay on/off) does this mean this same button will also pair my remote in the same way as the Sonoff RF?

Jason2866, 2018-12-30:
It is a receiver only, nothing is done in device. You get the IR Codes (only) via mqtt.
The magic has to be done in the Home Automation system or via rules in the device
For rules see wiki

betyar, 2018-12-30:
I understand that, that is what I was hoping with the module I have, I would use Node-Red to send a command to the relay when something was received from the module. From what I read elsewhere, this module sends simple on//off signals like a button or a switch since the pairing is done by the module itself. In fact, it has thee modes similar to a Sonoff 4CH Pro (momentary. inching, self-locking). So if I understand you correctly then, the module you mentioned would pass to mqtt anything it receives (since there is no pairing) and I would have to use Node-Red or a rule to filter what I get and use that  to toggle the relay. 

What I still don't understand, is if the module I have supposedly sends an on/off signal, why can't I configure Tasmota to receive this signal as if it was a button or switch and then act on that. For instance, in my setup GPIO2 and GPIO1 are free (GPIO0 is a button and GPIO3 is the relay). GPIO2 can be the RF receiver configured as a Button2 and GPI01 can be configured as a dummy relay (Relay2). I can then link the two relay states together. Of course this is all assuming that what I read elsewhere about the module I have is correct in the first place. 

Anyway, these are just thoughts to help me try and fully understand the process. Thanks very much for your help, I have already ordered the module you mentioned. I will keep this issue opened until I receive the module and tried it out to confirm that everything is OK which I am sure it will be. In the meantime if you or anyone else have any insights on how I can use the module I already have with Tasmota it would be great for a future project.

betyar, 2018-12-30:
This is what I tried at first and it didn't work. The console showed nothing.I tried both as a button and a switch. I just came across a post on a esp8266/arduino that perhaps the problem is that I need to connect the data pin using resistors like so: data -> 20K -> gpio2 ->10k -> GND.  Somehow I am not sure this is right but I might give it a try. Voltage can't be a problem because the module handles 3.3~5V input.

betyar, 2018-12-31:
Finally got it working. I simply added a 10K resistor from the data line to ground and configured the receiver as a switch (Switch1). The module can be programmed so that one button is on and the other is off, but they both use the same data line. You can also configure it so that the same button acts as toggle button.

Thanks for your help @Jason2866. The module I ordered will still come in handy for another project.

closed 2018-12-31 by betyar