arendst/Tasmota #4767: MQTT not working anymore

slam666, 2018-12-30:
Hi,

  I just did an upgrade from OpenHab 2.3 to 2.4 and also upgraded my Tasmota to 6.4.1 (Sonoff)

  Before it was working fine, now it seem that the MQTT command are not sent to my mosquitto broker.

  Topic is Lamp, so this is what I see on my Sonoff console:

17:50:55 MQT: tele/Lamp/STATE = {"Time":"2018-12-30T17:50:55","Uptime":"0T01:47:21","Vcc":3.557,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Home","BSSId":"A2:6C:AC:03:A0:7D","Channel":1,"RSSI":68}}
17:50:55 MQT: stat/Lamp/RESULT = {"POWER":"OFF"}
17:50:55 MQT: stat/Lamp/POWER = OFF

Any idea?

Jason2866, 2018-12-30:
Hi, it is a OpenHab issue. Wrong place, anyway....
Did the upgrade from 2.3. to 2.4 too. Had the same problem.
Try too uninstall mqtt binding and install it right afterwards.
After that all worked again.
Please close this issue even your problem isnt solved, OpenHab related!!

slam666, 2018-12-30:
18:49:28 MQT: stat/Tree/RESULT = {"FullTopic":"%prefix%/%topic%/"}
18:50:18 MQT: stat/Tree/RESULT = {"Topic":"Tree"}
setopion59=0 and setoption19=0
18:48:03 MQT: stat/Tree/RESULT = {"POWER":"ON"}
18:48:03 MQT: stat/Tree/POWER = ON
Anything else I can check for troubleshooting?

Jason2866, 2018-12-31:
Take a look in wiki. There are tipps for migration too
https://github.com/arendst/Sonoff-Tasmota/wiki/openHAB

slam666, 2018-12-31:
My Topic is Tree
so can anyone confirm if this is what I should see in MQTT:
stat/Tree/RESULT {"POWER":"ON"}
stat/Tree/POWER ON
stat/Tree/RESULT {"POWER":"OFF"}
stat/Tree/POWER OFF
Thanks.

andrethomas2, 2019-01-01:
Please note that GitHub Issues are for reporting bugs and requesting additional features for Tasmota firmware only.

As this appears to be an openHAB issue please see the wiki as advised by @Jason2866 and further direct your issues to the openHAB community.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-01-01 by andrethomas2

Rojer7084, 2020-08-08:
I try to change the "PubSubClient.h"   --->   from 256  to 1200    becuase if the MQTT package with "Enegry" shall be more  > 256 bytes   .  Please try it.   

// MQTT_MAX_PACKET_SIZE : Maximum packet size. Override with setBufferSize(). 
#ifndef MQTT_MAX_PACKET_SIZE
#define MQTT_MAX_PACKET_SIZE 1200
#endif