arendst/Tasmota #4769: Compiling error

stefanoMancon, 2018-12-30:
invalid conversion from 'char*' to 'const uint8_t* {aka const unsigned char*}' [-fpermissive]

Jason2866, 2018-12-30:
Your setup has to be faulty.
Latest dev version compiles without any error.
Since you ignored the needed issue template no help can be provided

andrethomas2, 2018-12-31:
Please note that GitHub Issues are for reporting bugs and requesting additional features.

Please search the Wiki for the information you require or visit Discord Chat where general questions and answers are handled.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-12-31 by andrethomas2