arendst/Tasmota #4775: Problem with Sonoff TH16

1Topf, 2018-12-31:
Hello everybody :o),
I have a new TH16 (with DS18B20-sensor) and flashed it with Tasmota 6.2.1. But I can`t see the temperature. When I choose "AM2301", I see "nan°C". Any ideas?
Many greetings and a Guten Rutsch ins neue Jahr :-D,
Andi

[image]

[image]

[image]

wongnam, 2018-12-31:
Chose module DS18x20 (04).
Should use the latest FW for stable and more functionality. 

Happy New Year!!!

1Topf, 2019-01-01:
Happy new year. :-)
Update done without success.

With sensor DS18x20 (04):

[image]

With sensor AM2301 (02):

[image]

my_user_config.h:

[image]

Perhaps sensor defective?

Jason2866, 2019-01-01:
Plug isnt correct in socket!
Try it with a little more force

wongnam, 2019-01-01:
a TH10/16 that had this problem that the jack was very loose, need force press the 2.5mm male jack to get better contact.
#2168

ascillato2, 2019-01-02:
Hi,

Please, try the solution proposed and report back if it is working for you. Thanks

closed 2019-01-02 by ascillato2