arendst/Tasmota #4776: Issue when build: low free memory, OTA options?

puppetmaster886, 2018-12-31:
Hi, I make a mistake when building the firmware and now I found that all my devices had very low remaining memory, and I need to update it by ota because the devices are **inside the wall**.

The devices have the 6.2.1, this memory:
Flash Chip Id	1327198
Flash Size	1024kB
Program Flash Size	1024kB
Program Size	447kB
**Free Program Space	44kB
Free Memory	15kB**

I am a developer with relative experience on IOT, do you think there is any possibility that I could modify the minimal version (or any other code out there) to enable ota functionality on the small free memory that my devices have?

andrethomas, 2018-12-31:
Try command
`reset 5`
It will erase all your settings and the unused flash portion but will keep your wifi settings.

After complete, connect again and see if space has free'd up.

arendst, 2019-01-01:
Your initial memory model is wrong. You did not use 1M0 linker but probably 512M0 as your total program memory is only about 512k.

Solution: reflash via serial using the correct memory model (linker 1M0)

closed 2019-01-01 by andrethomas2