arendst/Tasmota #4777: S31 not displaying energy usage

CleverCompiler, 2018-12-31:
I just flashed tasmota to my first sonoff device. After selecting the 41 sonoff S31 module.  I have turned the log up to level 4. I do see additional logging but I don't see any errors in the console log.

This is my first experience with sonoff and tasmota. Any help is appreciated.

CleverCompiler, 2018-12-31:
6.4.1 BTW

meingraham, 2018-12-31:
Be sure serial logging is disabled and the baud rate is set to 4800 (this should happen when you select S31 as the module, but confirm).

With WebLog 4 on as you note, do you see any serial data being logged?

meingraham, 2018-12-31:
After setting the module to S31, double ensure that `SerialLog` is set to `0`. Perhaps a power cycle for good measure. If you still get no serial data in the console with WebLog 4, then you might have a problem with the flash and need to re-do it. If you have to re-flash, I recommend you do it hard-wired for this go around. And, when you do have it hooked up, erash the flash first before uploading TASMOTA.

I'd use a latest development branch - either self-compiled or pre-build from http://thehackbox.org/tasmota/

Once you get the module operating properly, the you should be able to perform future firmware upgrades via OTA.

Jason2866, 2019-01-01:
Are you sure that you have a S31 with energy monitoring?
There is a S31 lite which looks identic and this version has only wifi switch function.

CleverCompiler, 2019-01-01:
Thanks Jason2866 I didn't know the lite version existed but that's what I have. I've ordered the full version s31.

closed 2019-01-01 by CleverCompiler

CleverCompiler, 2019-01-01:
Thanks meingraham for your ideas. Apparently I bought the lite version which doesn't include power monitoring. I've ordered the correct version now.

Thanks again.