arendst/Tasmota #4779: Sonoff Pow & R2 - Suggestion to include support for Electricity rates and Billing Cycle

ctripodi, 2018-12-31:
Hello @arendst , Thak you for this gtreat project!
This post is related to a suggestion to include support for Electricity rates and Billing Cycle for the modules that support energy metter like Sonoff POW & POW R2.
Maybe this can be implemented using some new commands in the Console. 
For example the cost by kWh and the day for the Start Billing cycle. Moreover the Total Energy Usage between Billing cycles 
Thank you

arendst, 2019-01-01:
That would be a good candidate for your Home Automation System.

As billing is sometimes based on time of day (low and high price) this would mean a lot of stored parameters also depending on rate changes.

closed 2019-01-01 by arendst

chimcen, 2020-04-07:
Any news here? This would be great on all other devices, too (e.g. Gosund SP111).
If the result could be shown on mainpage of tasmota this would be awesome...