arendst/Tasmota #4780: Sonoff POW v2.0 flash succeeds but afterwards not working

riston, 2019-01-01:
I managed to flash Sonoff POW v2.0 with Tasmota binaries but the issue is that it now doesn't work. I am not sure wether the chip is some how broken or not.

I started with the precompiled binaries v6 and v5 and flashing(espeasy tools) succeeded but on device restart nothing actually booted. After this I tried compiling the source with Arduino IDE and reaced to same state.

Went through all the similar issues but didn't find anything useful https://github.com/arendst/Sonoff-Tasmota/issues/1203. Tried different upload speeds it will succeed with 115200 and starts to throw error messages with 9600.
 
For flashing I was using similar usb-ttl https://www.aliexpress.com/item/1PCS-CH340G-Serial-Converter-USB-2-0-To-TTL-6PIN-Module-for-PRO-mini-Instead-of/32674198611.html

- [x] Development/Compiler/Upload tools used : Arduino IDE and also precompiled binaries
- [x] Hardware used : Sonff POW v2.0
- [x] If a pre-compiled release or development binary was used, which one? : Tested latest v6.x and v5.x
- [x] You have tried latest release or development binaries? : Yes
- [x] Provide the output of command ``status 0`` : No output

Output to Termite or in Arduino monitor terminal when restarted.
```
[00]sd[00]lÜŸ<[00]„là<[03][0C][04][0C]Œ[04]lä[04]c|‚[03]ä[13]›;“cl[04]c„ûgnŸdogœãä[0C]c[1C]x„‡l;l{lxóoà[18][03][0C][04]Ã[0C]$[0C]„œ[0C][04][0C]#[04]gã|[03]dä[0C][04]‡#Äònoï[00]$Œl`[03][1B][13]og[04]l [03][0F][02]n{‡“›g[04][0C]c[04]œ[0F]l[07]s›’'[0C][0C]c[0C]œ[07]dÜ[03]ŒœœŒà€ä[0E]$üƒoœ[03]
```

Jason2866, 2019-01-01:
You wont get log infos on serial because it,is used for communication with power metering chip.

andrethomas2, 2019-01-01:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-01-01 by andrethomas2

riston, 2019-01-01:
Issue was closed so quickly, @Jason2866 can you be more specific, so after flashing I receive garbage information from power metering chip right? This would only happen when I have chosen the module type as noted from this page(https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-Pow-and-Pow-R2)? If I can't use serial connection what should I use, the web connection doesn't work?

Jason2866, 2019-01-01:
Use esptool as flashing tool (see wiki). Do a flash erase with esptool. After this flash firmware with esptool.
If Esptool finish with success it will work! If not your hardware or your usb serial adapter is faulty.
The usb serial adapter you have is a good one!
Disconnect adapter and connect POW to mains. It should work. 
To be honest i never look at serial terminal! just flash and use...

riston, 2019-01-03:
Managed to get Sonoff POW working, the issue was related too low current provided by USB hub. Also, make sure the flash tool does a content hash check which esptool does and the MD5 hashes didn't match. I used Raspberry PI with an external 3.3v power supply to flash the Sonoff which succeeded.