arendst/Tasmota #4781: Tasmota 6.4.1 DNS Resolve not work

m0rph2k, 2019-01-01:
Hi

After tasmota update from 6.3.0 to 6.4.1 my Sonoff TH16 can not resolve my MQTT Server FQDN. IP Adresse works fine. Downgrade to 6.3.0 and the name resolve works

Anyone have the same problems ??

m0rph2k, 2019-01-01:
With a older version 6.2.1 the name resolve works. With 6.4.0 not

andrethomas, 2019-01-01:
@m0rph2k Do you mean mDNS?

andrethomas, 2019-01-01:
The .local TLD is reserved for mDNS so unless you're explicitly using mDNS it will fail. It worked on core 2.3.0 because core 2.3.0 wrongfully allows failover to conventional DNS when mDNS broadcast requests failed.

Please see issue https://github.com/arendst/Sonoff-Tasmota/issues/4670 for more information as this was previously discussed.

closed 2019-01-01 by andrethomas2