arendst/Tasmota #4783: Wrong Switch Status reporting

GianCann, 2019-01-02:
**Describe the bug**
Command status 8 (or status 10) report a status of the switch than it's not really of the hardware status

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [X] _Searched the problem in issues and in the wiki_
- [X] _Hardware used_ : Az-Delivery [D1 Mini](https://www.az-delivery.de/collections/wifi-module/products/d1-mini?ls=en)
- [X] _Development/Compiler/Upload tools used_ : OTA Update
- [X] _You have tried latest release or development binaries?_ : Yes
- [X] _Provide the output of command_``status 0`` :
```
**STATUS 0 OUTPUT HERE - DO NOT DELETE THE MARKERS ABOVE AND BELOW THIS LINE**
8:56:24 CMD: status 0
08:56:24 SRC: WebConsole from 212.210.1.34
08:56:24 RSL: Group 0, Index 1, Command STATUS, Data 0
08:56:24 MQT: stat/TEST/STATUS = {"Status":{"Module":18,"FriendlyName":["EspTest1","EspTest2"],"Topic":"TEST","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"","SwitchMode":[0,1,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
08:56:24 MQT: stat/TEST/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:07:47","StartupUTC":"2019-01-02T07:48:37","Sleep":50,"BootCount":123,"SaveCount":889,"SaveAddress":"FA000"}}
08:56:24 MQT: stat/TEST/STATUS2 = {"StatusFWR":{"Version":"6.4.1(sonoff)","BuildDateTime":"2018-12-24T14:41:00","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
08:56:24 MQT: stat/TEST/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":3,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["BelkinWiFi",""],"TelePeriod":300,"SetOption":["1000C009","558180C0","80000008"]}}
08:56:24 MQT: stat/TEST/STATUS4 = {"StatusMEM":{"ProgramSize":521,"Free":480,"Heap":17,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
08:56:24 MQT: stat/TEST/STATUS5 = {"StatusNET":{"Hostname":"ESP8266-DEV","IPAddress":"192.168.1.113","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"CC:50:E3:17:CE:6D","Webserver":2,"WifiConfig":3}}
08:56:24 MQT: stat/TEST/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.111","MqttPort":1883,"MqttClientMask":"Display","MqttClient":"Display","MqttUser":"GianCann","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
08:56:24 MQT: stat/TEST/STATUS7 = {"StatusTIM":{"UTC":"Wed Jan 02 07:56:24 2019","Local":"Wed Jan 02 08:56:24 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:43","Sunset":"17:04"}}
08:56:24 MQT: stat/TEST/STATUS10 = {"StatusSNS":{"Time":"2019-01-02T08:56:24","Switch1":"ON","Switch2":"ON"}}
08:56:24 MQT: stat/TEST/STATUS11 = {"StatusSTS":{"Time":"2019-01-02T08:56:24","Uptime":"0T00:07:47","Vcc":3.072,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"BelkinWiFi","BSSId":"98:DE:D0:E8:8D:19","Channel":1,"RSSI":100}}}
```

**To Reproduce**
ESP8266 D1 Mini board with Tasmota 6.4.1
Configure as Generic (18)
Put Switch1 on D2 (GPIO4)
Put Switch2 on D1 (GPIO5)
Hardware status of the pin GPIO4 - GPIO5: OPEN (not connected to GND)
Write in the console 'status 8' (or 'status 10'): resulted Switch are ON but Switch are effectlty OFF
```
09:03:40 RSL: Group 0, Index 1, Command STATUS, Data 8
09:03:40 MQT: stat/TEST/STATUS8 = {"StatusSNS":{"Time":"2019-01-02T09:03:40","Switch1":"ON","Switch2":"ON"}}
```

**Expected behavior**
```
09:03:40 RSL: Group 0, Index 1, Command STATUS, Data 8
09:03:40 MQT: stat/TEST/STATUS8 = {"StatusSNS":{"Time":"2019-01-02T09:03:40","Switch1":"OFF","Switch2":"OFF"}}
```

**Additional context**
I would like to use D1 Mini also as multiple 'Binary sensor', reading the status of some Switch not linked to any relay/output. The status can be read at any time

arendst, 2019-01-02:
Nice finding.

You may also find that the text is ON and not SWITCH CLOSED.

So there is more to it...

Having an open GPIO with pull-up enabled as is the default in Tasmota means the signal is High which is noted as ON. The same for a GPIO connected to Gnd which will have a Low signal and is noted as OFF. It has nothing to do with the actually state of the controlled relay.

To make matters worse you can select the ``switchstate`` to report the inverse in which case the ON changes to OFF and reverse.

That solves your issue and hopefully broadens your horizon too ;-)

GianCann, 2019-01-02:
Thank you @arendst for the explanation.
Now, i've closed Switch2 to Gnd:
`10:52:01 MQT: tele/TEST/SENSOR = {"Time":"2019-01-02T10:52:01","Switch1":"ON","Switch2":"OFF"}`

OFF are for "connected" to Gnd

Also if i setting Switch2 as Switch2**n**, the status is the same:
11:04:04 MQT: tele/TEST/SENSOR = {"Time":"2019-01-02T11:04:04","Switch1":"ON","Switch2":"OFF"}

There is another way to read OPEN/CLOSE status of the switch?

arendst, 2019-01-02:
Yes. Look in wiki for command ``switchmode`` and it's many options.

closed 2019-01-02 by ascillato2