arendst/Tasmota #4784: Pin Configuration for Generic Module In Code Behind

manikdutta, 2019-01-02:
Hi sir,
I want pin configuration for the generic module (NodeMcuESP8266) in code behind i.e., in the Tasmota firmware not from the browser. If someone guides me on how to achieve this from code will be very helpful. The pin configuration is shown in the picture below.
[image]

arendst, 2019-01-02:
Learn to use commands ``gpio`` and ``gpios`` to be used by serial, http, webconsole and MQTT. See wiki > commands.

manikdutta, 2019-01-02:
Through commands, the goal will be achieved but I want this through the code

arendst, 2019-01-02:
Find function GpioInit and feed your userconfig file defines.

ascillato2, 2019-01-02:
@manikdutta 

Hi, 

Sorry, but It is not supported "by default" so you need to modify the code to achieve what you want. Search for the function gpioinit in Tasmota code. Thanks.

closed 2019-01-02 by ascillato2