arendst/Tasmota #4785: 2,9" EPaper Display not working

helmar74, 2019-01-02:
Hi,

I connected a 2.9" EPAper Display to Wemos Mini:
Here's how I connected:

WEMOS <-> DISPLAY
3,3 <-> VCC
GND <-> GND
D7 <-> DIN
D5 <-> CLK
D2 <-> CS

The Disply is set to 3-wire SPI with jumper-

Here's my Setting in Tasmota Webinterface:

[image]

My Problem is, that I can't set Displaymodel to 5:
`12:37:46 CMD: displaymodel 5
12:37:46 MQT: stat/sonoffStandard/RESULT = {"DisplayModel":0}
12:37:49 CMD: displaymodel
12:37:49 MQT: stat/sonoffStandard/RESULT = {"DisplayModel":0}`

When I enter 5 as Displaymodel, always 0 is shown in Webinterface.

What do I make wrong here?
Do I have to enter a Adress?

Thanks a lot?

**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Development/Compiler/Upload tools used :
- [X] Hardware used : 
- [X] If a pre-compiled release or development binary was used, which one? :
- [X] You have tried latest release or development binaries? :
- [X] Provide the output of command ``status 0`` :
```
2:33:49 CMD: status 0
12:33:49 MQT: stat/sonoffStandard/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"sonoffStandard","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
12:33:49 MQT: stat/sonoffStandard/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffStandardGruppe","OtaUrl":"http://192.168.1.50:8080/static/firmware/sonoffStandard.ino.bin","RestartReason":"Software/System restart","Uptime":"0T00:02:51","StartupUTC":"2019-01-02T11:30:58","Sleep":50,"BootCount":6,"SaveCount":12,"SaveAddress":"F8000"}}
12:33:49 MQT: stat/sonoffStandard/STATUS2 = {"StatusFWR":{"Version":"6.4.1(sonoff)","BuildDateTime":"2019.01.02 12:15:52","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
12:33:49 MQT: stat/sonoffStandard/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["XXXXXXX","sonoffap"],"TelePeriod":300,"SetOption":["00008009","558180C0","00000000"]}}
12:33:50 MQT: stat/sonoffStandard/STATUS4 = {"StatusMEM":{"ProgramSize":574,"Free":428,"Heap":14,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000407","0FFAE79C","00038FA0","23BE1FCE","00003BC0"]}}
12:33:50 MQT: stat/sonoffStandard/STATUS5 = {"StatusNET":{"Hostname":"sonoffStandard-6547","IPAddress":"192.168.1.124","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.2","Mac":"84:F3:EB:3B:79:93","Webserver":2,"WifiConfig":4}}
12:33:50 MQT: stat/sonoffStandard/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.50","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_3B7993","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
12:33:50 MQT: stat/sonoffStandard/STATUS7 = {"StatusTIM":{"UTC":"Wed Jan 02 11:33:50 2019","Local":"Wed Jan 02 12:33:50 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99,"Sunrise":"08:01","Sunset":"16:24"}}
12:33:50 MQT: stat/sonoffStandard/STATUS10 = {"StatusSNS":{"Time":"2019-01-02T12:33:50"}}
12:33:50 MQT: stat/sonoffStandard/STATUS11 = {"StatusSTS":{"Time":"2019-01-02T12:33:50","Uptime":"0T00:02:52","Vcc":2.980,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"XXXXXXX","BSSId":"9C:C7:A6:F7:72:A2","Channel":6,"RSSI":82}}}
```

**(Please, remember to close the issue when the problem has been addressed)**

helmar74, 2019-01-02:
Now I changed settings to:
[image]

Now I get:
`12:59:00 CMD: displaymodel 5
12:59:00 MQT: stat/sonoffStandard/RESULT = {"DisplayModel":4}
12:59:04 CMD: displaymodel
12:59:04 MQT: stat/sonoffStandard/RESULT = {"DisplayModel":4}
12:59:09 CMD: displayaddress
12:59:09 MQT: stat/sonoffStandard/RESULT = {"DisplayAddress1":113}
12:59:48 CMD: display
12:59:48 MQT: stat/sonoffStandard/RESULT = {"Display":{"Model":4,"Mode":0,"Dimmer":1,"Size":1,"Font":1,"Rotate":0,"Refresh":2,"Cols":[16,8],"Rows":2}}`

But still no Text output with "Displaytext BLABLA"

Jason2866, 2019-01-02:
DisplayMode 0?
DisplayModel is 4!! 
#define USE_DISPLAY_EPAPER_29 // [DisplayModel 5] Enable e-paper 2.9 inch display (+19k code)

helmar74, 2019-01-02:
Why can't I set to Model5?
`13:15:43 CMD: **displaymodel 5**
13:15:43 MQT: stat/sonoffStandard/RESULT = {"DisplayModel":**4**}`

helmar74, 2019-01-02:
Here's my current setting:
[image]

Here's how I connected:

WEMOS <-> DISPLAY
3,3 <-> VCC
GND <-> GND
D7 <-> DIN
D5 <-> CLK
D2 <-> CS

Is it correct connected?

Jason2866, 2019-01-02:
For display(s) i still recommend the gemu fork....
[image]

helmar74, 2019-01-02:
You have connected a E-Paper Display on GPIO 12-14?

Jason2866, 2019-01-02:
yes, i use gemu fork. He has implemented software SPI

helmar74, 2019-01-02:
OK, thanks. But Software SPI is also implemented in Tasmota 6.4.1...

Jason2866, 2019-01-02:
Displaymodel 5
Displayaddress 113
Displaymode 0

I dont know if the ePaper works with latest orig. Tasmota!

helmar74, 2019-01-02:
I tried to compile gemu fork, but i get this error:
`Arduino: 1.8.8 (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, Flash, ck, 26 MHz, 40MHz, DOUT, 1M (no SPIFFS), 2, v2 Higher Bandwidth, Disabled, None, Only Sketch, 115200"

U:\02_daten\27_Programmierung\04_Arduino_ESP8266\06_Projekte_8266_ESPxx\13_SONOFF_Tasmota\05_sonoff_EPaper\sonoffEPaper\support.ino: In function 'void GetFeatures()':

support:943:23: error: 'val' was not declared in this scope

   if (GPIO_SSPI_CS == val) { return true; }

                       ^

In file included from D:\Users\Helmut\Desktop\arduino-1.8.8_Tasmota_242_Gemu\portable\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/Arduino.h:29:0,

                 from sketch\sonoffEPaper.ino.cpp:1:

support:943:37: error: return-statement with a value, in function returning 'void' [-fpermissive]

   if (GPIO_SSPI_CS == val) { return true; }

                                     ^

support:944:25: error: 'val' was not declared in this scope

   if (GPIO_SSPI_MOSI == val) { return true; }

                         ^

In file included from D:\Users\Helmut\Desktop\arduino-1.8.8_Tasmota_242_Gemu\portable\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/Arduino.h:29:0,

                 from sketch\sonoffEPaper.ino.cpp:1:

support:944:39: error: return-statement with a value, in function returning 'void' [-fpermissive]

   if (GPIO_SSPI_MOSI == val) { return true; }

                                       ^

support:945:25: error: 'val' was not declared in this scope

   if (GPIO_SSPI_SCLK == val) { return true; }

                         ^

In file included from D:\Users\Helmut\Desktop\arduino-1.8.8_Tasmota_242_Gemu\portable\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/Arduino.h:29:0,

                 from sketch\sonoffEPaper.ino.cpp:1:

support:945:39: error: return-statement with a value, in function returning 'void' [-fpermissive]

   if (GPIO_SSPI_SCLK == val) { return true; }

                                       ^

support:948:26: error: 'val' was not declared in this scope

   if (GPIO_COLDISP_BL == val) { return true; }

                          ^

In file included from D:\Users\Helmut\Desktop\arduino-1.8.8_Tasmota_242_Gemu\portable\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/Arduino.h:29:0,

                 from sketch\sonoffEPaper.ino.cpp:1:

support:948:40: error: return-statement with a value, in function returning 'void' [-fpermissive]

   if (GPIO_COLDISP_BL == val) { return true; }

                                        ^

U:\02_daten\27_Programmierung\04_Arduino_ESP8266\06_Projekte_8266_ESPxx\13_SONOFF_Tasmota\05_sonoff_EPaper\sonoffEPaper\xplg_ws2812.ino: At global scope:

xplg_ws2812:88:18: error: conflicting declaration 'uint8_t kRepeat [5]'

 uint8_t kRepeat[5] = {

                  ^

In file included from U:\02_daten\27_Programmierung\04_Arduino_ESP8266\06_Projekte_8266_ESPxx\13_SONOFF_Tasmota\05_sonoff_EPaper\sonoffEPaper\xdrv_05_irremote.ino:74:0:

D:\Users\Helmut\Desktop\arduino-1.8.8_Tasmota_242_Gemu\sketchbook\libraries\IRremoteESP8266-2.5.2.03\src/IRrecv.h:26:16: error: 'kRepeat' has a previous declaration as 'const uint64_t kRepeat'

 const uint64_t kRepeat = UINT64_MAX;

                ^

U:\02_daten\27_Programmierung\04_Arduino_ESP8266\06_Projekte_8266_ESPxx\13_SONOFF_Tasmota\05_sonoff_EPaper\sonoffEPaper\xplg_ws2812.ino: In function 'void Ws2812Gradient(uint8_t)':

xplg_ws2812:225:48: error: invalid types 'const uint64_t {aka const long long unsigned int}[uint8_t {aka unsigned char}]' for array subscript

   uint8_t repeat = kRepeat[Settings.light_width];  // number of scheme.count per ledcount

                                                ^

xplg_ws2812:235:37: error: invalid types 'const uint64_t {aka const long long unsigned int}[uint8_t {aka unsigned char}]' for array subscript

     if (kRepeat[Settings.light_width] > 1) {

                                     ^

Mehrere Bibliotheken wurden für "PubSubClient.h" gefunden
 Benutzt: D:\Users\Helmut\Desktop\arduino-1.8.8_Tasmota_242_Gemu\sketchbook\libraries\PubSubClient-EspEasy-2.6.09
 Nicht benutzt: D:\Users\Helmut\Desktop\arduino-1.8.8_Tasmota_242_Gemu\sketchbook\libraries\PubSubClient-2.6.09
Mehrere Bibliotheken wurden für "IRremoteESP8266.h" gefunden
 Benutzt: D:\Users\Helmut\Desktop\arduino-1.8.8_Tasmota_242_Gemu\sketchbook\libraries\IRremoteESP8266-2.5.2.03
 Nicht benutzt: D:\Users\Helmut\Desktop\arduino-1.8.8_Tasmota_242_Gemu\sketchbook\libraries\IRremoteESP8266-2.2.1.02
Mehrere Bibliotheken wurden für "TasmotaSerial.h" gefunden
 Benutzt: D:\Users\Helmut\Desktop\arduino-1.8.8_Tasmota_242_Gemu\sketchbook\libraries\TasmotaSerial-2.2.0
 Nicht benutzt: D:\Users\Helmut\Desktop\arduino-1.8.8_Tasmota_242_Gemu\sketchbook\libraries\TasmotaSerial-2.1.0
 Nicht benutzt: D:\Users\Helmut\Desktop\arduino-1.8.8_Tasmota_242_Gemu\sketchbook\libraries\TasmotaSerial-2.0.0
exit status 1
'val' was not declared in this scope

Dieser Bericht wäre detaillierter, wenn die Option
"Ausführliche Ausgabe während der Kompilierung"
in Datei -> Voreinstellungen aktiviert wäre.
`

helmar74, 2019-01-02:
Tried again with original tasmota. I can only select Displaymodel 1-4, but not Displaymodel 5.
Here's my_user_config.h:
`
  #define USE_DISPLAY                            // Add I2C Display Support (+2k code)
    #define USE_DISPLAY_MODES1TO5                // Enable display mode 1 to 5 in addition to mode 0
    #define USE_DISPLAY_LCD                      // [DisplayModel 1] Enable Lcd display (I2C addresses 0x27 and 0x3F) (+6k code)
    #define USE_DISPLAY_SSD1306                  // [DisplayModel 2] Enable SSD1306 Oled 128x64 display (I2C addresses 0x3C and 0x3D) (+16k code)
    #define USE_DISPLAY_MATRIX                   // [DisplayModel 3] Enable 8x8 Matrix display (I2C adresseses see below) (+11k code)
      #define MTX_ADDRESS1     0x71              // [DisplayAddress1] I2C address of first 8x8 matrix module
      #define MTX_ADDRESS2     0x74              // [DisplayAddress2] I2C address of second 8x8 matrix module
      #define MTX_ADDRESS3     0x75              // [DisplayAddress3] I2C address of third 8x8 matrix module
      #define MTX_ADDRESS4     0x72              // [DisplayAddress4] I2C address of fourth 8x8 matrix module
      #define MTX_ADDRESS5     0x73              // [DisplayAddress5] I2C address of fifth 8x8 matrix module
      #define MTX_ADDRESS6     0x76              // [DisplayAddress6] I2C address of sixth 8x8 matrix module
      #define MTX_ADDRESS7     0x00              // [DisplayAddress7] I2C address of seventh 8x8 matrix module
      #define MTX_ADDRESS8     0x00              // [DisplayAddress8] I2C address of eigth 8x8 matrix module

#endif  // USE_I2C

// -- SPI sensors ---------------------------------
#define USE_SPI                                  // Hardware SPI using GPIO12(MISO), GPIO13(MOSI) and GPIO14(CLK) in addition to two user selectable GPIOs(CS and DC)

#ifdef USE_SPI
  #ifndef USE_DISPLAY
  #define USE_DISPLAY                            // Add SPI Display support for 320x240 and 480x320 TFT
  #endif
    #define USE_DISPLAY_ILI9341                  // [DisplayModel 4] Enable ILI9341 Tft 480x320 display (+19k code)
    #define USE_DISPLAY_EPAPER_29                 // [DisplayModel 5] Enable e-paper 2.9 inch display (+19k code)
#endif  // USE_SPI`

... and here my current settings:
[image]

arendst, 2019-01-02:
Ah, I see I forgot to change the define USE_DISPLAY_EPAPER in file xdsp_05_epaper_29.ino at line 22.

If you compile yourself you might want to try to change it in 
```
#ifdef USE_DISPLAY_EPAPER_29
```
and try again.

helmar74, 2019-01-02:
Hi, thanks. In which line should I enter the code?

Jason2866, 2019-01-02:
As Theo wrote line 22

helmar74, 2019-01-02:
Hi, I changed it, now I can select displaymodel 5.

... But I still can't display a text.
`16:55:49 CMD: displaytext [I]
16:55:52 MQT: stat/sonoffStandard/RESULT = {"DisplayText":"[I]"}
16:55:56 CMD: displaytext [z]
16:55:56 MQT: stat/sonoffStandard/RESULT = {"DisplayText":"[z]"}
16:56:04 CMD: displaytext asfdsfdsf
16:56:04 MQT: stat/sonoffStandard/RESULT = {"DisplayText":"asfdsfdsf"}`

[image]

Do I make something wrong?

helmar74, 2019-01-02:
After a "hard" reboot (power off and on) it works!

These are my current settings:
[image]

Thanks a lot for you help ;-)

arendst, 2019-01-02:
Congratulations!!

Alas my e-paper seems to be broken. It uses a lot of current while only power is connected so I guess it's dead. Just ordered a new one... 

I'll update the driver file with the correct define so it will work from dev 6.4.1.4 and up.

closed 2019-01-02 by arendst

mraczka, 2019-01-22:
When I set up WeMos like @helmar74 my device reboots and deletes all pin configuration. Any idea why?
23:07:01 CMD: status 0
23:07:01 MQT: stat/eink/STATUS = {"Status":{"Module":18,"FriendlyName":["eink"],"Topic":"eink","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
23:07:01 MQT: stat/eink/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Exception","Uptime":"0T00:07:04","StartupUTC":"2019-01-22T21:59:57","Sleep":50,"BootCount":34,"SaveCount":81,"SaveAddress":"FA000"}}
23:07:01 MQT: stat/eink/STATUS2 = {"StatusFWR":{"Version":"6.4.1.10(1fe1839-display)","BuildDateTime":"2019-01-22T18:03:01","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
23:07:01 MQT: stat/eink/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Aurora Borealis",""],"TelePeriod":300,"SetOption":["00008009","558180C0","00000000"]}}
23:07:01 MQT: stat/eink/STATUS4 = {"StatusMEM":{"ProgramSize":520,"Free":480,"Heap":14,"ProgramFlashSize":1024,"FlashSize":16384,"FlashChipId":"1840EF","FlashMode":3,"Features":["00000809","0FAAA59C","00038FA0","23B617CA","00003BC0"]}}
23:07:01 MQT: stat/eink/STATUS5 = {"StatusNET":{"Hostname":"eink-0666","IPAddress":"192.168.1.222","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"A0:20:A6:04:02:9A","Webserver":2,"WifiConfig":5}}
23:07:01 MQT: stat/eink/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.1","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_04029A","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
23:07:01 MQT: stat/eink/STATUS7 = {"StatusTIM":{"UTC":"Tue Jan 22 22:07:01 2019","Local":"Tue Jan 22 23:07:01 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:32","Sunset":"17:31"}}
23:07:01 MQT: stat/eink/STATUS10 = {"StatusSNS":{"Time":"2019-01-22T23:07:01"}}
23:07:01 MQT: stat/eink/STATUS11 = {"StatusSTS":{"Time":"2019-01-22T23:07:01","Uptime":"0T00:07:04","Vcc":3.026,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"","BSSId":"EC:08:6B:27:31:29","Channel":13,"RSSI":72}}}

Jason2866, 2019-01-23:
Delete flash with esptool erase flash. After that flash Tasmota again

mraczka, 2019-01-25:
Nothing changed. Any other idea?

arendst, 2019-01-25:
You loose configs as your device does many reboots in a short period of time. If you try to get an e-paper display functional this is most likely caused by miscommunication between your device and the display.

Depending on the kind of e-paper you have you might need more control over the e-paper interface.

My first e-paper display contained a 5V to 3V3 converter for both power and data lines. It only needed the 3-wire SPI interface (you need to re-solder the jumper on the back of the display) to function.

My seconds e-paper display does not have this voltage converter and to function it needs to control both DC and RST signal line too. I had to connect them also to as relays configured outputs. After a power on of the device you will have to tweak with these relays to get the display going. My current config looks like this:

[image]

After power on nothing happens. After a restart the display is updated.

Not nice but this leaves you to document a better way.

tremichl, 2019-09-08:
Same problem here as mraczka commented. When I set up WeMos D1 mini like @helmar74 my device reboots and deletes all pin configuration. As all SSPI ports in this case are outputs only, it should be possible to save SPI setup with no display connected? I have tried that, same problem. Erasing flash doesn't help either. Used pre-compiled version 6.6.0(Display) with Core 2_5_2
Any ideas?

```
11:22:37 CMD: status 0
11:22:37 MQT: stat/e-Paper/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"e-Paper","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
11:22:37 MQT: stat/e-Paper/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-display.bin","RestartReason":"Exception","Uptime":"0T00:09:41","StartupUTC":"2019-09-08T10:12:56","Sleep":50,"CfgHolder":4617,"BootCount":8,"SaveCount":20,"SaveAddress":"F8000"}}
11:22:37 MQT: stat/e-Paper/STATUS2 = {"StatusFWR":{"Version":"6.6.0(display)","BuildDateTime":"2019-07-06T14:34:56","Boot":31,"Core":"2_5_2","SDK":"2.2.1(cfd48f3)"}}
11:22:37 MQT: stat/e-Paper/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["TNet01",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100060000000000000000000000","00000000"]}}
11:22:37 MQT: stat/e-Paper/STATUS4 = {"StatusMEM":{"ProgramSize":546,"Free":456,"Heap":19,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","0FAAA39E","00178FA0","22B617C1","010028C0"]}}
11:22:37 MQT: stat/e-Paper/STATUS5 = {"StatusNET":{"Hostname":"e-Paper-1339","IPAddress":"10.152.52.160","Gateway":"10.152.52.254","Subnetmask":"255.255.255.0","DNSServer":"10.152.52.254","Mac":"BC:DD:C2:24:85:3B","Webserver":2,"WifiConfig":4}}
11:22:37 MQT: stat/e-Paper/STATUS6 = {"StatusMQT":{"MqttHost":"10.152.52.111","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_24853B","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
11:22:37 MQT: stat/e-Paper/STATUS7 = {"StatusTIM":{"UTC":"Sun Sep 08 10:22:37 2019","Local":"Sun Sep 08 11:22:37 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"06:16","Sunset":"19:19"}}
11:22:37 MQT: stat/e-Paper/STATUS10 = {"StatusSNS":{"Time":"2019-09-08T11:22:37"}}
11:22:37 MQT: stat/e-Paper/STATUS11 = {"StatusSTS":{"Time":"2019-09-08T11:22:37","Uptime":"0T00:09:41","Heap":19,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"TNet01","BSSId":"44:4E:6D:17:3A:FE","Channel":1,"RSSI":68,"LinkCount":1,"Downtime":"0T00:00:04"}}}
```

hardlog, 2021-03-14:
Hi,

I also have the problem that I cannot use the display.
I have a waeshare 2.9 V2.1 and bridge it in 3-wire SPI mode. On the ESP8266 (wemos D1) I have GND, 3.3V, CS on D8 (GPIO15), DIN (MOSI) on D7 (GPIO13) and CLK on D5 (GPIO14) as described above. Precompiled Tasmota-display.bin (9.3.1).
Configured:

[image]

Displaymodel 5,  Displaymode 0

Nothing comes of "displaytext blahblah".

What am I doing wrong.

Greetings Florian

ascillato, 2021-03-14:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md)

Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

mercator2012, 2021-03-24:
> Hi,
>
> I also have the problem that I cannot use the display.
> I have a waeshare 2.9 V2.1 and bridge it in 3-wire SPI mode. On the ESP8266 (wemos D1) I have GND, 3.3V, CS on D8 (GPIO15), DIN (MOSI) on [...]

Hello @hardlog,

did you manage to get the e-paper display working?

Markus

Jason2866, 2021-03-25:
You have to use SSPI

hardlog, 2021-03-25:
Hello,

 

no, i did not.

 

 

 
 

Gesendet: Mittwoch, 24. März 2021 um 21:44 Uhr

GryKyo, 2021-04-01:
I am also having trouble getting a Waveshare 2.9" EPD to operate using just about every combination of ideas suggested here!

This is from the console on boot:
`00:00:00.050 CFG: Loaded from flash at F9, Count 187
00:00:00.055 QPC: Count 1
00:00:00.058 SPI: Hardware using GPIO14(CLK), GPIO13(MOSI) and GPIO12(MISO)
00:00:00.019 DSP: E-Paper 2.9
00:00:00.022 Project tasmota Display Wemos D1 Version 9.3.1(display)-2_7_4_9(2021-03-31T00:15:11)
00:00:00.659 WIF: Connecting to AP1 Private Network Channel 7 BSSId 10:7B:EF:F3:D2:78 in mode 11n as display-1505...`

to me this is suggesting that all is well for hardware SPI.  The connections are correct MOSI to DIN looks good and when I select EPAPER29 CS for the chip select line, the firmware automatically eables [displaymodel: 5] as displayed in the console output?

Any more ideas?  FYI: I have also tried various combinations of SSPI also...

Thanks

Garry

karoCB, 2023-01-10:
lilygo epaper 2.9 T5 V2.2 - nach 2 Jahren noch keine Lösung für einen Betrieb mit Tasmota - oder ?

Jason2866, 2023-01-10:
There is an update. Don't know if other 2.9 are working with.
https://github.com/arendst/Tasmota/pull/17627

karoCB, 2023-01-13:
without parameter column A - no function possible

[https://forum.creationx.de/forum/index.php?thread/3494-esp32-tasmota-template-f%C3%BCr-liliygo-t5-v2-2-2-9-e-paper-display/&postID=39510#post39510]