arendst/Tasmota #4786: MQTT connection not possible when using FQDNs suddenly

feldi740, 2019-01-02:
Hi,
I received a couple of Shelly2 s to test with, so I updated the Tasmota package in my platformio from version 6.2.1.1-2_3_0 to version 6.4.1.1-2_4_2 because I wanted the ability to have the power meter been supported on the Shelly2. So far it was going well but after installing Tasmota on the Shelly2, I saw it isn't able to connect to my MQTT server via its FQDN - it only works when using ip address.
I was able to reproduce the issue on different devices (Wemos D1 mini, Sonoff 4CH Pro and Sonoff Basic) than the Shelly one with the firmware mentioned above. All existing devices running the 6.2.1.1 firmware are using the FQDN for MQTT server w/o any issue. The DNS server is configured correctly as well as all other networking related settings.
Any idea?

Thanks,
-Thomas

Jason2866, 2019-01-02:
Please fill out issue template

ascillato2, 2019-01-02:
There is no change in Tasmota code for DNS between 6.2.1 and 6.4.1

Please, check your config or provide the requested template or use the Tasmota Support Chat. Thanks

closed 2019-01-02 by ascillato2