arendst/Tasmota #4787: Question regarding documentation

Pewibe, 2019-01-02:
I like to understand the possible selections of Menu "configuration / configure module" fields "GPIOx". I searched a lot but didn't find explanations. 
In case there is nothing written please explain the following. If wanted I put your answers together as nice text and add it to the wiki. What I understand are the large list of sensor names as special configuration of the associated GPIOx. What I don't understand:
1. A number like (XXX) at text end - is it just an index or is there a deeper meaning?
2. Now to the expressions BUTTON, SWITCH, RELAY, LED, PWM, COUNTER:
2a. What is the meaning of small letter "i" or "n" in selections like "BUTTON2n", "RELAY4i"?
2b. Is it correct that BUTTON, SWITCH, COUNTER are inputs, RELAY, LED, PWM outputs?
3. Are there some general thoughts of the constructor of this GPIOx system? For example I don't understand why I need in addition to a SWITCH3 / BUTTON3 input for an external signal another definion of i.e. a RELAY3 to get a message a the input event?
Thanks in advance!

Jason2866, 2019-01-02:
1. Can you give a example or link to it?
2a. "i" is for inverted "n" NO intern pullup resistor 
2b. Yes
3. read wiki https://github.com/arendst/Sonoff-Tasmota/wiki/Buttons-and-Switches
                   https://github.com/arendst/Sonoff-Tasmota/wiki/Control-other-devices

ascillato2, 2019-01-02:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-01-02 by ascillato2

Pewibe, 2019-01-02:
@Jason2866 
1. "Can you give a example or link to it?" --> [image]
2. "Read ..." --> I read these and the information is fine, but explains only the deeper usage of SWITCH / BUTTON. No information regarding my questions.
3. What about an answer / tip related to zhis item?