arendst/Tasmota #4789: Random Reboot Sonoff T1/T2 when MQTT + Hue Bridge enabled

ymurawski, 2019-01-02:
Hi, i have faced out a problem when MQTT (iobroker with Sonoff adapter) and Hue Bridge Emulation in Tasmota is enabled.

I have many random reboots of my devices.(Sonoff T1/T2/T3 and Sonoff Basic) 

When i deactivate Hue Bridge Emulation everything works perfectly with IoBroker.

In attachment you see some pictures. 
"Decken Licht HWR" is using Hue Emulation and "Decken Licht Kinderzimmer" is not using Hue Emulation
There you can see the runtime of both devices.

What i tried so far is:
-complete flash delete
-new Firmware 6.4.1.4 with Core 2.5 and 2.4.2
-sleep 1
-sleep 0
-sleep 50

See attachment for Error Logs and screenshots

[image]
[image]
[image]
[image]

ascillato, 2019-01-02:
Hi, 

Can you please complete the troubleshooting template? 

When you have the reboot? At anytime or doing something in particular?

Have you tried core 2.3.0?

Can you self-compile? If so, try to disable mdns discovery

ymurawski, 2019-01-02:
Hi, 
i already tried with core 2.3, same problem.
the reboots are in random time. even when i dont do anything. 
I didnt to self compile before.

ascillato2, 2019-01-02:
Ok, Can you please complete the troubleshooting template?

ascillato2, 2019-01-02:
Are you using the hue bridge emulation for Alexa?

Jason2866, 2019-01-02:
Dont use special characters in SSID

ymurawski, 2019-01-02:
> Dont use special characters in SSID

But does this cause the problem ? When Hue Bridge is off the i have no dissconnects.

ascillato, 2019-01-02:
Alexa uses the cloud a lot ;)

The troubleshooting template is there when you open a new issue. You have either opened a blank issue or you have deleted the text.

ascillato, 2019-01-02:
Please, try to change your wifi ssid as Jason suggested. No special characters neither spaces

Jason2866, 2019-01-02:
You asked for help, so it is your choice...

ymurawski, 2019-01-02:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

**Describe the bug**
_A clear and concise description of what the bug is._

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x] _Searched the problem in issues and in the wiki_
- [x] _Hardware used_ : Sonoff Touch, Sonoff Basic
- [x] _Development/Compiler/Upload tools used_ : esptool
- [x] _If a pre-compiled release or development binary was used, which one?_ : 6.4.1.4 with 2.5+2.4.2
- [x] _You have tried latest release or development binaries?_ : yes
- [x] _Provide the output of command_``status 0`` :

```
18:40:47 MQT: stat/sonoff/STATUS = {"Status":{"Module":28,"FriendlyName":["Decken Licht HWR"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":8,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
```

**To Reproduce**
_Steps to reproduce the behavior:_
-Enable Sonoff adapter in IoBroker
-Enable Hue Bridge and MQTT in Tasmota

**Expected behavior**
_A clear and concise description of what you expected to happen._
- a stable connection with IoBroker and Hue Bridge for Alexa

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Additional context**
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2019-01-02:
Status 0 is incomplete.

Please do a status 0 in the console

ymurawski, 2019-01-02:
19:45:58 MQT: stat/sonoff/STATUS = {"Status":{"Module":28,"FriendlyName":["Decken Licht HWR"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":8,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
19:45:58 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://192.168.1.200:32769/data/firmwares/sonoff.bin","RestartReason":"Software Watchdog","Uptime":"0T00:01:02","StartupUTC":"2019-01-02T18:44:56","Sleep":1,"BootCount":129,"SaveCount":392,"SaveAddress":"F6000"}}
19:45:58 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.4.1.4(447ec62-sonoff)","BuildDateTime":"2019-01-01T15:14:58","Boot":31,"Core":"2_5_0_BETA2","SDK":"3.0.0-dev(c0f7b44)"}}
19:45:58 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["WIFI Home IoT","WIFI @ Home / IoT"],"TelePeriod":300,"SetOption":["00008009","55C180C0","00000000"]}}
19:45:58 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":546,"Free":456,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
19:45:58 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"Sonoff-Hauswirtsch.-1ch","IPAddress":"192.168.1.75","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"60:01:94:AC:CA:87","Webserver":2,"WifiConfig":4}}
19:45:58 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.205","MqttPort":1883,"MqttClientMask":"Sonoff-Hauswirtschaft-1ch","MqttClient":"Sonoff-Hauswirtschaft-1ch","MqttUser":"admin","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
19:45:59 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Wed Jan 02 18:45:58 2019","Local":"Wed Jan 02 19:45:58 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:43","Sunset":"17:04"}}
19:45:59 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-01-02T19:45:59"}}
19:45:59 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-01-02T19:45:59","Uptime":"0T00:01:03","Vcc":3.499,"SleepMode":"Dynamic","Sleep":1,"LoadAvg":999,"POWER":"OFF","Wifi":{"AP":1,"SSId":"WIFI Home IoT","BSSId":"CC:40:D0:08:CD:8D","Channel":9,"RSSI":92}}}

Jason2866, 2019-01-02:
The issue is IoBroker related
https://github.com/arendst/Sonoff-Tasmota/issues/4721

ymurawski, 2019-01-02:
So i opened a new WIFI called "WIFI Home IoT" but still the same.

ymurawski, 2019-01-02:
sonoff.0 | 2019-01-02 20:01:43.792 | info | Client [Sonoff-Hauswirtschaft-1ch] reconnected. Old secret 1546455645386_4734. New secret 1546455703792_7794
-- | -- | -- | --
sonoff.0 | 2019-01-02 20:00:45.386 | info | Client [Sonoff-Hauswirtschaft-1ch] connected with secret 1546455645386_4734
sonoff.0 | 2019-01-02 20:00:45.386 | info | Client [Sonoff-Hauswirtschaft-1ch] connection closed: Error: read ECONNRESET

ymurawski, 2019-01-02:
> The issue is IoBroker related
> #4721

Yes this was my Post but it has been closed and not reactivated

Jason2866, 2019-01-02:
Yes, because it is NOT a problem from Tasmota

ymurawski, 2019-01-02:
> Yes, because it is NOT a problem from Tasmota

Ok thanks for the information. What can i do next ? :(

Jason2866, 2019-01-02:
Contact Iobroker people for help.

ymurawski, 2019-01-02:
Ok i will try. I will post here if there is a solution

ascillato2, 2019-01-03:
There is something definetely wrong in your firmware ("LoadAvg":999) It is using full cpu (?) It is processing a lot of requests from your network?

ascillato, 2019-01-03:
I also use TasmoAdmin with the 5sec default and mine "LoadAvg":19

So, there is something in your network. To achieve that usage Tasmota is receiving a lot of requests. May be from mDNS.

Try the following. In the Tamota console type `weblog 4` to display more debug information and there you can see if there is something wrong with mDNS or other types of requests.

ascillato, 2019-01-03:
Ok, great. You have found the issue.

The adapter in iobroker is not working well and it is spamming your network.

If the sonoff adapter is not working and you are facing a lot of issues with iobroker, may be it is better to change to another Home Automation Software. There is a list of softwares with links and tutorials for Tasmota in the Tasmota wiki

ascillato2, 2019-01-03:
That is out of memory. Not enough RAM.

ascillato2, 2019-01-03:
The **epc1:0x4023e42e** is refering to an internal function of the core 2.5.0 (In the http://thehackbox.org/tasmo there is also a MAP file that is a text file with those numbers EPC and the functions related)

ymurawski, 2019-01-03:
Is that a serious problem?
You mean not enough RAM on my Sonoff device?!

ascillato2, 2019-01-03:
which brand of router do you have?

ascillato2, 2019-01-03:
Seems that your Sonoff is using all the available RAM due to the large amount of requests.

ascillato2, 2019-01-03:
Please, try the reduced Tasmota version: http://thehackbox.org/tasmota/020500/sonoff-basic.bin

ymurawski, 2019-01-03:
I use netgear Router because I need a 10gb/s Sfp Port for my network server

The Most common error that comes when the sonoff is restarting is RestartReason":"Software Watchdog"

ascillato, 2019-01-03:
Please, try the reduced Tasmota version: http://thehackbox.org/tasmota/020500/sonoff-basic.bin

That one doesn't have mDNS discovery and some sensors

ymurawski, 2019-01-03:
So far it stayes online. Lets See if this works. I will let it run for some hours and then Check it again

ascillato, 2019-01-03:
If it works, the issue is some device spamming in your network with mDNS requests.

ymurawski, 2019-01-03:
For now its running 20min without an error.
With sonoff.bin it restartet around after 5min.

But LoadAvg":999 is still the same. 

Maybe the hue Emulation uses that much?

ascillato, 2019-01-03:
The hue emulation is very light. It remains below 50.

The good thing is that now your device is not rebooting. You should check your network traffic for multicast protocols

ascillato2, 2019-01-03:
Closing this issue as it is not a Tasmota issue.

Anyway, please, keep us posted if you find the device that is making lots of multicast requests. That would be very useful for other users having the same trouble.

Also you can try to turn off multicast in your router or changing the multicast IGMP version in your router config to V3.

closed 2019-01-03 by ascillato2

ymurawski, 2019-01-03:
Hi, i found the Problem with my AVGLoad 999

It was just the command sleep 1

with sleep 50 it dropped instantly to avgload 19-20