arendst/Tasmota #4792: Sleep 0

slam666, 2019-01-03:
Sonoff basic with Tasmota 6.4.1
Set Sleep to 0 result in being unable to connect to web page and light is blinking.
Any way to fix this?  Really hope I dont have to reflash.

Einstein67, 2019-01-03:
Press the button for 40s. Then "sleep" should be 50 (default).

Jason2866, 2019-01-03:
@slam666 
You have probably a sonoff basic R2. This is the main reason for programming a
own Tasmota sleep function. Sleep isnt longer based on Espresiff core code.
The sonoff Basic R2 has hardware design issues. The power supply cant deliever
enough current when booting and the heat cooling design of the ESP is worse too...
The device failure results in that what you describe. No wifi connect, led blinking, reboots...

The 6.4.x version of Tasmota takes care of this. BUT only if sleep is enabled!
You may have luck with long button press, if not you can try (last chance before flash via serial)
to power the device for 3-5 seconds and unpower it again. Do this about 10 times.
Tasmota has a inbuild fallback function to use defaults if booting fails often in short times

closed 2019-01-03 by andrethomas2