arendst/Tasmota #4797: corrupted color values MQTT

egs2003, 2019-01-03:
If I transmit via MQTT the color (201,204,0) the console shows me {"POWER":"ON","Dimmer":80,"Color":"200,204,0","HSBColor":"61,100,80","Channel":[78,80,0]}

There is a digit failure

Tasmota 6.4.1

Jason2866, 2019-01-03:
Can confirm:
```
11:08:16 CMD: color FFFFFFFF
11:08:16 MQT: stat/sonoff-0CB8DA/RESULT = {"POWER":"ON","Dimmer":100,"Color":"FFFFFFFF","HSBColor":"0,0,100","Channel":[100,100,100,100]}
11:12:54 CMD: color 201,204,0
11:12:54 MQT: stat/sonoff-0CB8DA/RESULT = {"POWER":"ON","Dimmer":80,"Color":"200,204,0,0","HSBColor":"61,100,80","Channel":[78,80,0,0]}
11:13:15 CMD: color 255,255,255
11:13:15 MQT: stat/sonoff-0CB8DA/RESULT = {"POWER":"ON","Dimmer":100,"Color":"255,255,255,0","HSBColor":"0,0,100","Channel":[100,100,100,0]}
11:13:51 CMD: color 201,204,0
11:13:51 MQT: stat/sonoff-0CB8DA/RESULT = {"POWER":"ON","Dimmer":80,"Color":"200,204,0,0","HSBColor":"61,100,80","Channel":[78,80,0,0]}
```
Status 0:
```
11:15:26 CMD: status 0
11:15:26 MQT: stat/sonoff-0CB8DA/STATUS = {"Status":{"Module":27,"FriendlyName":["AiLight"],"Topic":"sonoff-0CB8DA","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
11:15:26 MQT: stat/sonoff-0CB8DA/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin","RestartReason":"External System","Uptime":"0T00:08:06","StartupUTC":"2019-01-03T10:07:20","Sleep":50,"BootCount":25,"SaveCount":560,"SaveAddress":"F5000"}}
11:15:26 MQT: stat/sonoff-0CB8DA/STATUS2 = {"StatusFWR":{"Version":"6.4.1.4(sonoff)","BuildDateTime":"2019-01-02T18:16:53","Boot":31,"Core":"STAGE","SDK":"3.0.0-dev(c0f7b44)"}}
11:15:26 MQT: stat/sonoff-0CB8DA/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Jason_Home_WLAN",""],"TelePeriod":300,"SetOption":["00028009","55C180C0","00000001"]}}
11:15:26 MQT: stat/sonoff-0CB8DA/STATUS4 = {"StatusMEM":{"ProgramSize":491,"Free":512,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashMode":3,"Features":["00000809","0F8AA784","040103A1","80041496","00002CC2"]}}
11:15:26 MQT: stat/sonoff-0CB8DA/STATUS5 = {"StatusNET":{"Hostname":"sonoff-0CB8DA-6362","IPAddress":"192.168.2.129","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"DC:4F:22:0C:B8:DA","Webserver":2,"WifiConfig":3}}
11:15:26 MQT: stat/sonoff-0CB8DA/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.114","MqttPort":1883,"MqttClientMask":"sonoff-0CB8DA","MqttClient":"sonoff-0CB8DA","MqttUser":"openhabian","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
11:15:26 MQT: stat/sonoff-0CB8DA/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 03 10:15:26 2019","Local":"Thu Jan 03 11:15:26 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99,"Sunrise":"08:05","Sunset":"16:28"}}
11:15:26 MQT: stat/sonoff-0CB8DA/STATUS10 = {"StatusSNS":{"Time":"2019-01-03T11:15:26"}}
11:15:26 MQT: stat/sonoff-0CB8DA/STATUS11 = {"StatusSTS":{"Time":"2019-01-03T11:15:26","Uptime":"0T00:08:06","Vcc":3.545,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":236,"POWER":"ON","Dimmer":80,"Color":"200,204,0,0","HSBColor":"61,100,80","Channel":[78,80,0,0],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","BSSId":"00:A0:57:1E:69:50","Channel":4,"RSSI":86}}}
```

arendst, 2019-01-03:
As designed. Search issues.

In short it is rounding from 255 to 100 to 255.

egs2003, 2019-01-03:
Why?