arendst/Tasmota #4798: Export Pow R2 measurements to an Excel sheet.

MPJK77, 2019-01-03:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)

I have a Sonoff Pow R2 flashed with Tasmota 6.3.0.
Is it possible to export my Pow R2 measurements to an Excel sheet?

andrethomas, 2019-01-03:
You can collect telemetry data using an HA system like node-red and create comma separated value files and ultimately excel data files from there. The detail on a per hour or per minute basis is not stored on the device itself so you need to handle this off-device.

MPJK77, 2019-01-03:
@andrethomas 
Thank you very much for your quick reply!

I am using Home Assistant (Hass.io) and I have integrated the sensor in there,
but I don't know how to create / extract the csv files from there.
Maybe this question needs to be in the HA topic?

andrethomas, 2019-01-03:
You would need to ask on the HA forums/Issues/Discord or some similar place.

closed 2019-01-03 by andrethomas2