arendst/Tasmota #4799: print out configuration

egs2003, 2019-01-03:
I want a command to see all configuration made

ascillato, 2019-01-03:
Hi,

Duplicated issue. 

There is already a an option to download all the config to a file and a script to decode it. 

Please, search in issues for more information. Thanks

ascillato2, 2019-01-03:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-01-03 by ascillato2

curzon01, 2019-01-04:
see [decode-config](https://github.com/arendst/Sonoff-Tasmota/blob/development/tools/decode-config.md) - it is not neccessary to previously save a config file, the tool is also able to decode the config directly from the device.

egs2003, 2019-01-04:
Danke, das hat mir geholfen!