arendst/Tasmota #4800: Sonoff Slampher RF, Can be flashed, but not functional

indiejoseph, 2019-01-03:
It has been successfully flashed with latest Tasmota firmware, after flashed, the LED not lighting and cannot find the connection on my router.

**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Development/Compiler/Upload tools used : Arduino IDE
- [x] Hardware used : Sonoff Slampher RF with USB serial
- [x] If a pre-compiled release or development binary was used, which one? : Compiled by Arduino IDE
- [x] You have tried latest release or development binaries? : Yes
- [x] Provide the output of command ``status 0`` : Cannot connect to WIFI
```
STATUS 0 OUTPUT HERE
```

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2019-01-03:
Hi,

- Have you disconnected GPIO0 from GND?
- Can you try the precompiled bins instead?
- Have you asked for help in the Tasmota Support Chat?

ascillato2, 2019-01-03:
Please, address this flashing issue to the Tasmota Support Chat. It is better and more dynamic Q&A in the Chat. Github issues are for bug reporting. Thanks.

closed 2019-01-03 by ascillato2