arendst/Tasmota #4801: Constant reset on Shelly1 when using button1n

subzero79, 2019-01-03:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

**Describe the bug**
When the shelly1 is configured as a generic module and set the switch to a button, constant reset at boot happens due to a long press (not intentional) 40 seconds after  start boot.
Configuration is GPIO04 -> relay1, GPIO05 -> button1n

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [ x] _Searched the problem in issues and in the wiki_:  yes similar issue is described here https://github.com/arendst/Sonoff-Tasmota/issues/1117 for a sonoff touch. Also there is some instability issues in the original support shelly thread regarding using button but with GPIO0 and GPIO3 
- [x ] _Hardware used_ : shelly1 v2
- [ x] _Development/Compiler/Upload tools used_ : Visual Studio(Linux)
- [ ] _If a pre-compiled release or development binary was used, which one?_ :
- [ x] _You have tried latest release or development binaries?_ : ? Yes, built from git
- [ x] _Provide the output of command_``status 0`` :

Status 0 as default (sonoff basic module)
```
23:10:26 MQT: sonoff/stat/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":1,"SwitchRetain":1,"SensorRetain":0,"PowerRetain":1}}
23:10:26 MQT: sonoff/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:00:30","StartupUTC":"2019-01-03T13:09:56","Sleep":50,"BootCount":2,"SaveCount":6,"SaveAddress":"F6000"}}
23:10:26 MQT: sonoff/stat/STATUS2 = {"StatusFWR":{"Version":"6.4.1.4(sonoff)","BuildDateTime":"2019-01-03T11:46:04","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
23:10:26 MQT: sonoff/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["sharktopus-IoT",""],"TelePeriod":300,"SetOption":["000A80E9","558180C0","00000200"]}}
23:10:26 MQT: sonoff/stat/STATUS4 = {"StatusMEM":{"ProgramSize":532,"Free":468,"Heap":22,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540EF","FlashMode":3,"Features":["00000809","0FDAEB94","240383A1","23B617CE","00003BC0"]}}
23:10:26 MQT: sonoff/stat/STATUS5 = {"StatusNET":{"Hostname":"sonoff-3160","IPAddress":"10.10.4.35","Gateway":"10.10.4.1","Subnetmask":"255.255.255.0","DNSServer":"10.10.4.1","Mac":"CC:50:E3:05:8C:58","Webserver":2,"WifiConfig":2}}
23:10:26 MQT: sonoff/stat/STATUS6 = {"StatusMQT":{"MqttHost":"10.10.0.152","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_058C58","MqttUser":"homeassistant","MqttType":2,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
23:10:26 MQT: sonoff/stat/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 03 13:10:26 2019","Local":"Thu Jan 03 23:10:26 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+10:00","Sunrise":"04:58","Sunset":"18:44"}}
23:10:26 MQT: sonoff/stat/STATUS10 = {"StatusSNS":{"Time":"2019-01-03T23:10:26"}}
23:10:26 MQT: sonoff/stat/STATUS11 = {"StatusSTS":{"Time":"2019-01-03T23:10:26","Uptime":"0T00:00:30","Vcc":3.194,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"sharktopus-IoT","BSSId":"82:2A:A8:5A:9F:F2","Channel":6,"RSSI":100}}}
```
Status 0 after setting the generic module with the conf described above

```
23:12:03 MQT: sonoff/stat/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":1,"SwitchRetain":1,"SensorRetain":0,"PowerRetain":1}}
23:12:03 MQT: sonoff/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:09","StartupUTC":"2019-01-03T13:11:54","Sleep":50,"BootCount":3,"SaveCount":11,"SaveAddress":"F9000"}}
23:12:03 MQT: sonoff/stat/STATUS2 = {"StatusFWR":{"Version":"6.4.1.4(sonoff)","BuildDateTime":"2019-01-03T11:46:04","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
23:12:03 MQT: sonoff/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["sharktopus-IoT",""],"TelePeriod":300,"SetOption":["000A80E9","558180C0","00000200"]}}
23:12:03 MQT: sonoff/stat/STATUS4 = {"StatusMEM":{"ProgramSize":532,"Free":468,"Heap":23,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540EF","FlashMode":3,"Features":["00000809","0FDAEB94","240383A1","23B617CE","00003BC0"]}}
23:12:03 MQT: sonoff/stat/STATUS5 = {"StatusNET":{"Hostname":"sonoff-3160","IPAddress":"10.10.4.35","Gateway":"10.10.4.1","Subnetmask":"255.255.255.0","DNSServer":"10.10.4.1","Mac":"CC:50:E3:05:8C:58","Webserver":2,"WifiConfig":2}}
23:12:03 MQT: sonoff/stat/STATUS6 = {"StatusMQT":{"MqttHost":"10.10.0.152","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_058C58","MqttUser":"homeassistant","MqttType":2,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
23:12:03 MQT: sonoff/stat/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 03 13:12:03 2019","Local":"Thu Jan 03 23:12:03 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+10:00","Sunrise":"04:58","Sunset":"18:44"}}
23:12:03 MQT: sonoff/stat/STATUS10 = {"StatusSNS":{"Time":"2019-01-03T23:12:03"}}
23:12:03 MQT: sonoff/stat/STATUS11 = {"StatusSTS":{"Time":"2019-01-03T23:12:03","Uptime":"0T00:00:09","Vcc":3.181,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"sharktopus-IoT","BSSId":"82:2A:A8:5A:9F:F2","Channel":6,"RSSI":100}}}

```

**To Reproduce**
Flash the shelly1 with tasmota firmware, configure it as generic module with GPIO04 as relay1 and GPIO5 as button1n, reboot happens, after 40-50 seconds the device resets to defaults and reboots back to sonoff basic module. All settings are lost.

**Expected behavior**
Configure as generic module set the switch input to work as button and have the module retain its settings

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Additional context**
This is the log taken from the serial console (without mains connected of course) and the same happens, but here i am able to catch the button press at boot and the reset at the end which is not possible to capture using the web console

``
00:00:00 SRC: Restart
00:00:00 Project sonoff Sonoff Version 6.4.1.4(sonoff)-2_4_2
00:00:00 APP: Button1 multi-press 1
```
https://pastebin.com/9Ssbigkp

**(Please, remember to close the issue when the problem has been addressed)**

Thanks very much

ascillato, 2019-01-03:
Hi,

Seems to be a bad fashing issue.

Please, erase all flash with esptool.py (as explained in the wiki) and flash again with the precompiled bins from development branch http://thehackbox.org/tasmota/

closed 2019-01-03 by ascillato2

ascillato, 2019-01-05:
Try `SetOption13 1` to disable the multipress feature to avoid your reset.

ascillato2, 2019-01-06:
https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#16-using-an-external-button-with-single-press---double-press-and-hold

subzero79, 2019-01-06:
This works well but connecting the press to gpio0 instead of the switch terminal.

mweinelt, 2020-02-15:
Sorry for negro-bumping this, but this is the thread I found by googling.

I had this problem for a while now on Shelly 1 and 1PM. What finally fixed the resets was setting `SetOption1 1`, which literally says it prevents inadvertent resets.