arendst/Tasmota #4803: MQTT: Channel<x> influences other channels

Gert05, 2019-01-03:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

**Describe the bug**
Issuing a channel command for one channel changes values for other channels.

Behaviour seen on H801.
Start with all channels 0.
Set 1st channel by "channel1 30" -> response includes "Channel":[28,0,0,0,0]
Set 2nd channel by "channel2 30" -> response includes "Channel":[27,28,0,0,0]

Obviously, channel1 should not be changed by a channel2-command.

Secondly, shouldn't a "channel1 30" lead to a "Channel":[30,0,0,0,0]?

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x] _Searched the problem in issues and in the wiki_
- [x] _Hardware used_ : H801
- [x] _Development/Compiler/Upload tools used_ : OTA 6.4.1
- [ ] _If a pre-compiled release or development binary was used, which one?_ :
- [x] _You have tried latest release or development binaries?_ : yes, latest release
- [ ] _Provide the output of command_``status 0`` :
```
17:36:34 CMD: status 0
17:36:34 MQT: stat/h801/STATUS = {"Status":{"Module":20,"FriendlyName":["Sonoff"],"Topic":"h801","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[1,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
17:36:34 MQT: stat/h801/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"External System","Uptime":"0T00:33:18","StartupUTC":"2019-01-03T16:03:16","Sleep":50,"BootCount":55,"SaveCount":812,"SaveAddress":"FB000"}}
17:36:34 MQT: stat/h801/STATUS2 = {"StatusFWR":{"Version":"6.4.1(sonoff)","BuildDateTime":"2018-12-24T14:41:00","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
17:36:34 MQT: stat/h801/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Maxnet",""],"TelePeriod":300,"SetOption":["00008009","558180C0","00000001"]}}
17:36:34 MQT: stat/h801/STATUS4 = {"StatusMEM":{"ProgramSize":521,"Free":480,"Heap":17,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440E0","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
17:36:34 MQT: stat/h801/STATUS5 = {"StatusNET":{"Hostname":"h801-5438","IPAddress":"192.168.178.185","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"89.101.251.229","Mac":"18:FE:34:99:55:3E","Webserver":2,"WifiConfig":2}}
17:36:34 MQT: stat/h801/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.4","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_99553E","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
17:36:34 MQT: stat/h801/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 03 16:36:34 2019","Local":"Thu Jan 03 17:36:34 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:43","Sunset":"17:05"}}
17:36:34 MQT: stat/h801/STATUS10 = {"StatusSNS":{"Time":"2019-01-03T17:36:34","Switch8":"ON"}}
17:36:34 MQT: stat/h801/STATUS11 = {"StatusSTS":{"Time":"2019-01-03T17:36:34","Uptime":"0T00:33:18","Vcc":3.510,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":60,"POWER":"ON","Dimmer":29,"Color":"4649000000","HSBColor":"62,100,29","Channel":[27,28,0,0,0],"CT":499,"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"Maxnet","BSSId":"78:8A:20:B7:B9:CC","Channel":2,"RSSI":100}}}
```

**To Reproduce**
See bug description

**Expected behavior**
See bug description

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Additional context**
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2019-01-03:
Hi,

Sorry,

Duplicated issue https://github.com/arendst/Sonoff-Tasmota/issues/4797

closed 2019-01-03 by ascillato2

Gert05, 2019-01-03:
Hi,
Agree with respect to the secondary issue, but not with respect to the main topic. 
If changing one channel leads to changes in the other four channels (slowly reducing values to zero...) the remark in #4797 “it’s a feature, not a bug” does not cover the essence of the problem. 
    	
    	

    	Outlook voor iOS downloaden