arendst/Tasmota #4804: switchtopic 3 still firing relay

joearkay, 2019-01-03:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Development/Compiler/Upload tools used : ESPTool
- [x] Hardware used : NodeMCU (ESP8266)
- [x] If a pre-compiled release or development binary was used, which one? : sonoff.bin
- [x] You have tried latest release or development binaries? : Yes
- [x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE

9:16:35 RSL: stat/sonoff/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"test","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"switch_trig","SwitchMode":[0,0,2,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
19:16:35 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:11:02","StartupUTC":"2019-01-03T18:05:33","Sleep":50,"BootCount":9,"SaveCount":68,"SaveAddress":"F8000"}}
19:16:35 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.4.1(sonoff)","BuildDateTime":"2018-12-24T14:41:00","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
19:16:35 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["HAL9000",""],"TelePeriod":300,"SetOption":["00008009","558180C0","00000000"]}}
19:16:35 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":521,"Free":480,"Heap":21,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640C8","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
19:16:35 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"Fan-heater-sonoff","IPAddress":"192.168.1.143","Gateway":"192.168.1.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.254","Mac":"DC:4F:22:0D:4F:8C","Webserver":2,"WifiConfig":4}}
19:16:35 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.1","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0D4F8C","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
19:16:35 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 03 18:16:35 2019","Local":"Thu Jan 03 19:16:35 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:43","Sunset":"17:05"}}
19:16:35 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-01-03T19:16:35","Switch3":"ON"}}
19:16:35 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-01-03T19:16:35","Uptime":"0T00:11:02","Vcc":3.021,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"HAL9000","BSSId":"FC:EC:DA:84:63:C2","Channel":6,"RSSI":100}}}
```

**(Please, remember to close the issue when the problem has been addressed)**

Hi all, 

I've had a good read and search through the forums, Google and the Github issues section but am still struggling with these. I'm doing something similar to what was described here:(https://github.com/arendst/Sonoff-Tasmota/issues/2036) whereby I want an input (D2, Switch3) to just publish an MQTT topic and not fire Relay1, on D1. 

I have set `switchtopic switch_trig` via the UI console, which successfully echos back `19:14:34 RSL: stat/sonoff/RESULT = {"SwitchTopic":"switch_trig"}` when I enter `switchtopic`. From my understanding, this should publish the status of the switch under `switch_trig` and in therefore NOT trigger any onboard relays. However, Relay1 on-board is still getting triggered - what am I doing wrong here?

ascillato, 2019-01-03:
Please, address this to the Tasmota Chat. There will be easier for Q&A

Seems that you have defined your switch as button or something wrong in your config.

ascillato2, 2019-01-03:
### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-01-03 by ascillato2

joearkay, 2019-01-04:
Just leaving the solution here after excellent help from @ascillato2 👍 

This was all done via the console on the UI. In the console type:
1) `rule1 1` (to enable rules)
2) `switchmode3 1`
3) `rule1 on switch3#state=1 do publish cmnd/switch_trig/switch3 ON endon on switch3#state=0 do publish cmnd/switch_trig/switch3 OFF endon` (to make your reed to "follow" the state of your reed - 1 when closed / 0 - when open)
4) `switchtopic 0` (as switchtopic usually overrides rules, so it needs to be disabled)

hoatienii, 2019-04-20:
> Just leaving the solution here after excellent help from @ascillato2 👍
>
> This was all done via the console on the UI. In the console type:
>
> 1. `rule1 1` (to enable rules)
> 2. `switchmode3 1`
> 3. `rule1 [...]

hi! i follow command you type, but nothing happen.
I have sonoff T1 US 3gang. i want to use button 3 to publish command cmnd/plug/switch3 ON if i press button again publish command cmnd/plug/switch3 OFF
Could you help me!