arendst/Tasmota #4805: When selecting DS18B20 in new basic, the temp is missing

seritos, 2019-01-03:
I have the new board of Sonoff basic. I have solder a DS18B20, however, when selecting it in GPIO2, the temperature is not displayed in the main page: 

[image]

[image]

Even is the sensor wouldn't be soldered, the temperature line should appear. isn't it?

Thanx

andrethomas, 2019-01-03:
> Even is the sensor wouldn't be soldered, the temperature line should appear. isn't it?

No

seritos, 2019-01-03:
I have removed the Ds18b20, added a DHT11 and it works fine. 

The temperature line does appear even though the sensor is not connected. Try it.

andrethomas, 2019-01-03:
> The temperature line does appear even though the sensor is not connected. Try it.

Once detected yes it will. But if not connected on power-up it will not.

Neil-M0NFI, 2019-01-03:
It works for me on GPIO2 with the new version of Sonoff Basic. Have you connected the 4K7 resistor between 3.3v and GPIO2 that the DS18B20 needs?

seritos, 2019-01-03:
No I haven't! 

Didn't know this was needed! Is this only with the Ds18b20?

seritos, 2019-01-03:
It works! Thanx much for your supoort! 

Just a couple of questions...

Is it posible to read the temperature in Homekit with 0,1 degrees definition instead of 0,5? 

Is there any possibility with tasmota to create, trough an app or similar, a graph with the historial temperature?  

Thanx!

andrethomas, 2019-01-03:
> Is it posible to read the temperature in Homekit with 0,1 degrees definition instead of 0,5?

If you use the correct sensor it can provide finer than 0.5deg resolution - LM75AD, for example, can provide up to 0.125 deg C resolution - Some users use the BMP/BME 180,280 and 680 sensors which are more reliable and consistent in that order.

> Is there any possibility with tasmota to create, trough an app or similar, a graph with the historical temperature?

All data can be transmitted via mqtt to a home automation system which can do this recording for you. There are many options available... OpenHab, HomeAssistant, Node-red etc

closed 2019-01-03 by andrethomas2