arendst/Tasmota #4806: Domotocz not working & no Domoticcz web Interface after flashing FW

chrom1, 2019-01-03:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Development/Compiler/Upload tools used :
- [ ] Hardware used : 
- [ ] If a pre-compiled release or development binary was used, which one? :
- [ ] You have tried latest release or development binaries? :
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**(Please, remember to close the issue when the problem has been addressed)**

chrom1, 2019-01-03:
Hello all, i flashed a OBI power plug with "Sonoff Basic" (Sonoff-Tasmota-6.4.1). The webinterface is working but no Domoticz is selectable; I set type to "18 Generic" and changed also GPIO settings as described here: [https://github.com/arendst/Sonoff-Tasmota/wiki/OBI-Socket-2]
In Konsole i can read

> 00:00:06 MQT: Verbindungsversuch...
> 00:00:06 MQT: verbunden

In my Domoticz log i dont find anything of the OBI plug
Any Idea?

chrom1, 2019-01-03:
here my STATUS 0 Output:
22:18:30 CMD: status 0
22:18:30 MQT: stat/sonoffOBI/STATUS = {"Status":{"Module":18,"FriendlyName":["SonoffOBI"],"Topic":"sonoffOBI","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
22:18:30 MQT: stat/sonoffOBI/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:04:12","StartupUTC":"2019-01-03T21:14:18","Sleep":50,"BootCount":16,"SaveCount":39,"SaveAddress":"F5000"}}
22:18:30 MQT: stat/sonoffOBI/STATUS2 = {"StatusFWR":{"Version":"6.4.1(basic)","BuildDateTime":"2019.01.03 21:44:49","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
22:18:30 MQT: stat/sonoffOBI/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Chromi",""],"TelePeriod":300,"SetOption":["00008009","558180C0","00000000"]}}
22:18:30 MQT: stat/sonoffOBI/STATUS4 = {"StatusMEM":{"ProgramSize":445,"Free":556,"Heap":26,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540A1","FlashMode":3,"Features":["00000407","0F082780","24008000","00000006","000001C0"]}}
22:18:30 MQT: stat/sonoffOBI/STATUS5 = {"StatusNET":{"Hostname":"sonoffOBI-0700","IPAddress":"xx.xx.xx.xx","Gateway":"xx.xx.xx.xx","Subnetmask":"255.255.255.0","DNSServer":"xx.xx.xx.xx","Mac":"80:7D:3A:16:82:BC","Webserver":2,"WifiConfig":4}}
22:18:30 MQT: stat/sonoffOBI/STATUS6 = {"StatusMQT":{"MqttHost":"xx.xx.xx.xx","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_1682BC","MqttUser":"pi","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
22:18:30 MQT: stat/sonoffOBI/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 03 21:18:30 2019","Local":"Thu Jan 03 22:18:30 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:43","Sunset":"17:05"}}
22:18:30 MQT: stat/sonoffOBI/STATUS10 = {"StatusSNS":{"Time":"2019-01-03T22:18:30"}}
22:18:30 MQT: stat/sonoffOBI/STATUS11 = {"StatusSTS":{"Time":"2019-01-03T22:18:30","Uptime":"0T00:04:12","Vcc":3.414,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Chromi","BSSId":"A4:52:6F:43:19:3E","Channel":11,"RSSI":60}}}

ascillato, 2019-01-03:
Hi,

Have you followed the wiki instructions? If so, which steps have you done? Which approach have you choose (manual or automatic configuration) 

As this issue is just configuration issue, please address this to the Tasmota Support Chat. Github issues are for bug reporting.

The chat is better to address this type of issues due to there is a lot of information missing. Have you configured mqtt? Have you follow the wiki? Have you use the automatic approach of the wiki or manual configuration for domoticz? Have you flash by ota or wire? Etc.

Thanks

closed 2019-01-03 by ascillato2