arendst/Tasmota #4807: Flash using raspberry pi 3B+ reboot issue

mikewebb70, 2019-01-04:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Development/Compiler/Upload tools used :
- [ ] Hardware used : 
- [ ] If a pre-compiled release or development binary was used, which one? :
- [ ] You have tried latest release or development binaries? :
- [ ] Provide the output of command ``status 0`` :
```
reboot
```
Following the instructions:
 https://github.com/arendst/Sonoff-Tasmota/wiki/Flash-Sonoff-using-Raspberry-Pi 
fails for me if I follow the wiring diagram on that page.

When I connect the 3V3 on the Sonoff basic to the RP3B+ to enter flash mode.  The RP3B+ reboots.  I tried 3 different RP3B+ along with different power supplies.

If I use an external 3V3 source (I used one from a nodemcu),  Everything else in the instructions works successfully.

I'm not sure if this a thing specific to the new 3B+ versions? but putting this out there.

Thank you for TASMOTA and all the quality documentation.

closed 2019-01-04 by mikewebb70

GianCann, 2019-01-04:
Suggestion: @ascillato2  can you add the link to the index of the documentation?

Morphy99, 2019-01-28:
Even in the linked blog post from Michel Deslierres which the wiki is based off mentions this:

> I often lost the SSH connection with the Raspberry Pi when I put the Sonoff into programming mode. The solution was to give the Sonoff an independent 3.3V power supply. 

So I would say it's best not to use the onboard 3.3V