arendst/Tasmota #4812: documentation needed for IRremote

aihysp, 2019-01-04:
there are no Docs related on how to send IR MQTT request 
if someone can please add a way to use this

Wiki refers us to [this](https://github.com/altelch/SonoffIR)
its great on hot to build and connect to the sonoff , but i am not getting hot to use it

curzon01, 2019-01-04:
Use [IRSend command](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#irremote) as described.

How to use commands in general see [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands)

ascillato2, 2019-01-04:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-01-04 by ascillato2

aihysp, 2019-01-04:
Many thanks, I will try!