arendst/Tasmota #4813: teleperiod 300, to many mqtt messages

1igeza, 2019-01-04:
HI, 
I have got a problem with my one sonoff pow. Teleperiod is set to 300, and in log i have:
`12:34:01 LOG: SerialLog 2, WebLog 2, SysLog 0, LogHost , LogPort 514, TelePeriod 300
12:34:16 MQT: rafal_pow/tele/SENSOR = {"Time":"2019-01-04T12:34:16","ENERGY":{"TotalStartTime":"2018-11-26T06:36:25","Total":0.153,"Yesterday":0.298,"Today":0.153,"Power":7,"ApparentPower":38,"ReactivePower":38,"Factor":0.18,"Voltage":247,"Current":0.155}}
12:34:20 MQT: rafal_pow/tele/SENSOR = {"Time":"2019-01-04T12:34:20","ENERGY":{"TotalStartTime":"2018-11-26T06:36:25","Total":0.153,"Yesterday":0.298,"Today":0.153,"Power":3,"ApparentPower":33,"ReactivePower":33,"Factor":0.09,"Voltage":245,"Current":0.135}}
12:36:49 MQT: rafal_pow/tele/SENSOR = {"Time":"2019-01-04T12:36:49","ENERGY":{"TotalStartTime":"2018-11-26T06:36:25","Total":0.153,"Yesterday":0.298,"Today":0.153,"Power":7,"ApparentPower":37,"ReactivePower":36,"Factor":0.18,"Voltage":247,"Current":0.149}}
12:36:54 MQT: rafal_pow/tele/SENSOR = {"Time":"2019-01-04T12:36:54","ENERGY":{"TotalStartTime":"2018-11-26T06:36:25","Total":0.153,"Yesterday":0.298,"Today":0.153,"Power":2,"ApparentPower":50,"ReactivePower":50,"Factor":0.04,"Voltage":247,"Current":0.203}}
12:36:56 MQT: rafal_pow/tele/SENSOR = {"Time":"2019-01-04T12:36:56","ENERGY":{"TotalStartTime":"2018-11-26T06:36:25","Total":0.153,"Yesterday":0.298,"Today":0.153,"Power":7,"ApparentPower":36,"ReactivePower":35,"Factor":0.19,"Voltage":247,"Current":0.144}}
12:37:02 MQT: rafal_pow/tele/SENSOR = {"Time":"2019-01-04T12:37:02","ENERGY":{"TotalStartTime":"2018-11-26T06:36:25","Total":0.153,"Yesterday":0.298,"Today":0.153,"Power":2,"ApparentPower":32,"ReactivePower":32,"Factor":0.07,"Voltage":244,"Current":0.132}}
12:37:20 MQT: rafal_pow/tele/SENSOR = {"Time":"2019-01-04T12:37:20","ENERGY":{"TotalStartTime":"2018-11-26T06:36:25","Total":0.153,"Yesterday":0.298,"Today":0.153,"Power":6,"ApparentPower":33,"ReactivePower":32,"Factor":0.18,"Voltage":247,"Current":0.132}}`

Also setopion19 is set to 0, but home assistant discover all date from this sonoff pow. I have 4 pcs of this, and all have tasmota 6.4.1.3, but problem is with only one, and I can find what could be a problem.

mike2nl, 2019-01-04:
@1igeza 
please fill in the requested information. You have seen it when you made these new issue.

ascillato, 2019-01-04:
You have set powerdelta. That is why you have all those reports. Please, search in issues and also in the wiki due this issue is duplicated and was addressed before

ascillato2, 2019-01-04:
Closing this issue as it is duplicated 

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-01-04 by ascillato2