arendst/Tasmota #4814: Sonoff 4CH Pro: Interlock

wewa00, 2019-01-04:
Hello,

I have a Sonoff 4Ch Pro flashed with Tasmota and also configured it as a "Sonoff 4 CH Pro" module. My problem is, that I only can turn one relay on. If I turn another relay on, the other relay is turned off.
Interlock ("SetOption14") is off.
How can I turn several relays on at the same time?

Best regards,
wewa00

Einstein67, 2019-01-04:
Set the hw-switches, as described in the wiki:

S6: 1
K5: all 1
K6: all 0

ascillato, 2019-01-04:
Please, read the wiki at sonoff ch4 pro and set your dip switches as explained there. Thanks

ascillato2, 2019-01-04:
Closing this issue as it has been duplicated. 

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-01-04 by ascillato2