arendst/Tasmota #4820: New Sonoff RF R2 power need to add gipo 2

desskyblue, 2019-01-04:
**Have you look for this feature in other issues and in the wiki?**

**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is._

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2019-01-04:
Ok, Thanks.

closed 2019-01-04 by ascillato2

ascillato, 2019-01-04:
PR https://github.com/arendst/Sonoff-Tasmota/pull/4821

ascillato, 2019-01-04:
Can you please post a picture of the board? Thanks.

desskyblue, 2019-01-04:
[image]
[image]
[image]
[image]

ascillato, 2019-01-04:
In an hour you will be able to download the new Tasmota version with this change from http://thehackbox.org/tasmota/

Thanks.

desskyblue, 2019-01-05:
Which version should I choose? I do not see gpio2 with sonoff rf.
http://thehackbox.org/tasmota/sonoff.bin

desskyblue, 2019-01-05:
Thanks for the fast action, it works very well.