arendst/Tasmota #4823: DMX

makejoint, 2019-01-05:
**Have you look for this feature in other issues and in the wiki?**
yes

**Is your feature request related to a problem? Please describe.**
Control DMX capable light (see lightmaXX LED NANO PAR) via MQTT on a Sonoff Basic (and OpenHab in my setup).

**Describe the solution you'd like**
Control via smartphone like a low-cost Philips HUE (with OpenHab).
 
**Describe alternatives you've considered**
Using Arduino for DMX and looking later at this project: http://log.munichmakerlab.de/post/136975240870/our-new-mqtt-dmx-controller-based-on-an

----------
Dear all,

I'm trying to understand how to modify @arendst code to implement DMX in my Sonoff network. 
I would like to use a Sonoff basic to turn on/off a light, which is capable of DMX. With Tasmota I'm able to use the relay of the Sonoff basic via MQTT. So far so good. 
When in ON state, I'm then wondering if I can use one of the GPIO (e.g. GPIO 4) to spit out continuously DMX commands to a MAX-485 (see https://playground.arduino.cc/DMX/DMXShield). What about adding a xdrv in the /sonoff/ and the libraries of munichmakerlab in /lib/ (see http://log.munichmakerlab.de/post/136975240870/our-new-mqtt-dmx-controller-based-on-an)?
Any suggestion? @arendst would it be possible?

Thank you all. Awesome work here!

ascillato, 2019-01-05:
Please, search in issues. This has been addressed before. Thanks.

makejoint, 2019-01-05:
@ascillato, referring to #3280: This commit seems lost in void. I cannot find a way to realise my requested feature. Do you have any idea on the setup?

makejoint, 2019-01-05:
Thank you for your answer. I still believe it's not a duplicate issue.
I'm searching for a solution able to control a relay as well. So something like turn on/off and if on, use DMX on a free GPIO. Not only, Tasmota would provide OTA, Web interface, MQTT and so on, all already coded. I believe the Tasmota framework and code structure is much better suited for this scope, since the DMX light is part of a home automation project, not continuously controlled by an external software. 
I am poorly able to implement munichmakerlab's code into Tasmota (e.g., in a basic firmware), since with my coding competencies it would take too long (without the right help).

I was wondering of using the 250ms task in the main loop, but then I'm missing the MQTT linkage. Do you have a suggestion/advices for implementation?

ascillato, 2019-01-05:
For adding this, it has to be done as a driver as the rest of sensors and protocols that Tasmota supports.

Let's see if there is someone interested on writing a non-blocking driver to use a light version of DMX on Tasmota. The main issue, as explained in the referenced issues, is that will not be possible to make a full version with real time effects and so on. Just couple of commands per second. That is why, for real time or demanding tasks it is better to use a dedicated specific hardware/software for that.

DMX has excelent devices to properly manage their protocol.

makejoint, 2019-01-05:
Agree: no real-time requirement on the DMX. I hope somebody with the right knowledge and time will want to help on this.

For my curiosity, is there any documentation I can read on the Tasmota driver building?

Thank you

ascillato, 2019-01-05:
Yes! In the wiki at https://github.com/arendst/Sonoff-Tasmota/wiki/Sensor-API

ascillato, 2019-01-05:
And then you can check any of the drivers files already in Tasmota as examples.

Remember that drivers have to be non-blocking.

makejoint, 2019-01-05:
Thx for your help! I'll try to improvise something 👍

ascillato, 2019-01-05:
Remember that you can ask for specific help here and in the Tasmota Support Chat. Any of those channels are valid for this.

thomasnordquist, 2019-03-25:
Would a pull-request be welcome? @ascillato
I got a scope, DMX equipment, and the experience.

DMX is a very primitive protocol and enables a lot of lighting extensions.
I, for instance, use 3 different magic-home controllers to share the load of my plant's lighting system.
At maximum load it can draw ~20Amps, 2.5Amps per channel, with DMX I could simply use a dedicated light controller like this from [LTECH](https://de.aliexpress.com/item/Neue-Led-DMX-PWM-Decoder-DC12-24V-eingang-3A-16CH-Max-48A-1152-W-ausgang-XLR/32959838760.html)

I don't expect this feature to consume a lot of program space, especially if one abuses existing i2c.
Ram could vary a LOT with implementation and supported channel count.

Why this would be awesome in Tasmota?
- Tasmota is very well maintained and even has a update server, reducing maintenance effort
- It would benefit a lot of tinkerers
- Tasmota already has MQTT-Support

ascillato, 2019-03-25:
Hi,

Pull Request are always welcome. :) 

Remember to change as few files as possible. If you can put all together in just one driver file would be better.

You can check the API that Tasmota uses for internal communications.

Also, add a compiling-time key in my_userconfig.h to add your new driver. Remember to let it disabled as default. 

If you have any question regarding the code, you can find us in the Tasmota support chat. Anyway, we can check the code when you make the PR. 

Thanks :+1:

makejoint, 2019-03-25:
Would be awesome! I can offer a (very simple) testing facility as well: a
sonoff basic, a max485 based module, a light with DMX IN/OUT, a lot of time
and happiness.

I can help with documentation as well if needed.

Thank you all!

arnaud-aub, 2021-04-14:
bonjour, je cherche depuis un certain temps une façon de contrôler mes lumières DMX via wifi, j'ai des Sonoff, des carte MAX-485 (puce RS485) et des prises DMX, comment puis-je faire pour les contrôler ? Avez vous un schéma électronique, et un programme ? Merci, bonne journée.

makejoint, 2021-04-15:
Hi Arnaud,

If I read correctly, you're searching for a schematic? Check the posts
above, since some folks in Munich implemented DMX on a ESP + MAX device.
I did not follow up, since it seems that DMX takes too much resources on a
Sonoff and requires a scheduler different from the one implemented in
Tasmota.

Have a great day