arendst/Tasmota #4824: Sonoff Slampher v2 Random Power ON/OFF

thematica, 2019-01-05:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Development/Compiler/Upload tools used : esptool.py
- [x] Hardware used : Raspberry Pi 2B
- [x] If a pre-compiled release or development binary was used, which one? : release binary 6.4.1
- [x] You have tried latest release or development binaries? : yes
- [x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
03:45:01 CMD: status 0
03:45:01 SRC: WebConsole from 192.168.7.50
03:45:01 RSL: Received Topic /status, Data Size 1, Data 0
03:45:01 RSL: Group 0, Index 1, Command STATUS, Data 0
03:45:01 MQT: stat/sonoff2/STATUS = {"Status":{"Module":9,"FriendlyName":["Sonoff"],"Topic":"sonoff2","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
03:45:01 MQT: stat/sonoff2/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:08:12","StartupUTC":"2019-01-05T02:36:49","Sleep":50,"BootCount":12,"SaveCount":67,"SaveAddress":"F8000"}}
03:45:01 MQT: stat/sonoff2/STATUS2 = {"StatusFWR":{"Version":"6.4.1(sonoff)","BuildDateTime":"2018-12-24T14:41:00","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
03:45:01 MQT: stat/sonoff2/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Gear 2nd",""],"TelePeriod":300,"SetOption":["00008009","558180C0","00000000"]}}
03:45:01 MQT: stat/sonoff2/STATUS4 = {"StatusMEM":{"ProgramSize":521,"Free":480,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
03:45:01 MQT: stat/sonoff2/STATUS5 = {"StatusNET":{"Hostname":"sonoff2-6698","IPAddress":"192.168.7.149","Gateway":"192.168.7.31","Subnetmask":"255.255.255.0","DNSServer":"192.168.7.31","Mac":"60:01:94:A3:3A:2A","Webserver":2,"WifiConfig":4}}
03:45:01 MQT: stat/sonoff2/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.7.5","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_A33A2A","MqttUser":"homeassistant","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
03:45:01 MQT: stat/sonoff2/STATUS7 = {"StatusTIM":{"UTC":"Sat Jan 05 02:45:01 2019","Local":"Sat Jan 05 03:45:01 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:43","Sunset":"17:07"}}
03:45:01 MQT: stat/sonoff2/STATUS10 = {"StatusSNS":{"Time":"2019-01-05T03:45:01"}}
03:45:01 MQT: stat/sonoff2/STATUS11 = {"StatusSTS":{"Time":"2019-01-05T03:45:01","Uptime":"0T00:08:12","Vcc":3.514,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Gear 2nd","BSSId":"F0:B4:29:1B:42:AD","Channel":10,"RSSI":82}}}
03:45:09 WIF: Checking connection...
03:45:09 WIF: Connected
```

**(Please, remember to close the issue when the problem has been addressed)**

sample log
`00:00:00 CFG: Loaded from flash at F4, Count 48
00:00:00 SRC: Restart
00:00:00 Project sonoff Sonoff Version 6.4.1(sonoff)-2_4_2
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 Gear 2nd in mode 11N as sonoff2-6698...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Connected
00:00:04 DNS: Initialized
00:00:04 HTP: Web server active on sonoff2-6698.local with IP address 192.168.7.149
00:00:04 APP: (UTC) Sat Jan 05 01:02:28 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
02:02:29 RSL: tele/sonoff2/UPTIME = {"Time":"2019-01-05T02:02:29","Uptime":"0T00:00:05"}
02:02:30 MQT: Attempting connection...
02:02:35 MQT: Connect failed to 192.168.7.5:1883, rc -2. Retry in 10 sec
02:02:35 HTP: Main Menu
02:02:39 APP: Boot Count 8
02:02:39 CFG: Saved to flash at FB, Count 49, Bytes 3584
02:02:42 HTP: Console
02:02:45 MQT: Attempting connection...
02:02:45 MQT: Connected
02:02:45 MQT: tele/sonoff2/LWT = Online (retained)
02:02:45 MQT: cmnd/sonoff2/POWER = 
02:02:45 MQT: Subscribe to cmnd/sonoff2/#
02:02:45 MQT: Subscribe to cmnd/sonoffs/#
02:02:45 MQT: Subscribe to cmnd/DVES_A33A2A_fb/#
02:02:45 MQT: tele/sonoff2/INFO1 = {"Module":"Slampher","Version":"6.4.1(sonoff)","FallbackTopic":"cmnd/DVES_A33A2A_fb/","GroupTopic":"sonoffs"}
02:02:45 MQT: tele/sonoff2/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff2-6698","IPAddress":"192.168.7.149"}
02:02:45 MQT: tele/sonoff2/INFO3 = {"RestartReason":"Software/System restart"}
02:02:45 MQT: stat/sonoff2/RESULT = {"POWER":"OFF"}
02:02:45 MQT: stat/sonoff2/POWER = OFF
02:02:46 SRC: MQTT
02:02:46 RSL: Group 0, Index 1, Command POWER, Data OFF
02:02:46 MQT: stat/sonoff2/RESULT = {"POWER":"OFF"}
02:02:46 MQT: stat/sonoff2/POWER = OFF
02:02:54 MQT: tele/sonoff2/STATE = {"Time":"2019-01-05T02:02:54","Uptime":"0T00:00:30","Vcc":3.514,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Gear 2nd","BSSId":"F0:B4:29:1B:42:AD","Channel":10,"RSSI":56}}
02:05:30 SRC: MQTT
02:05:30 RSL: Group 0, Index 1, Command POWER, Data ON
02:05:30 MQT: stat/sonoff2/RESULT = {"POWER":"ON"}
02:05:30 MQT: stat/sonoff2/POWER = ON
02:05:31 CFG: Saved to flash at FA, Count 50, Bytes 3584
02:05:33 SRC: MQTT
02:05:33 RSL: Group 0, Index 1, Command POWER, Data OFF
02:05:33 MQT: stat/sonoff2/RESULT = {"POWER":"OFF"}
02:05:33 MQT: stat/sonoff2/POWER = OFF
02:05:34 CFG: Saved to flash at F9, Count 51, Bytes 3584
02:07:54 MQT: tele/sonoff2/STATE = {"Time":"2019-01-05T02:07:54","Uptime":"0T00:05:30","Vcc":3.524,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Gear 2nd","BSSId":"F0:B4:29:1B:42:AD","Channel":10,"RSSI":74}}
02:09:39 APP: Button1 multi-press 1
02:09:39 SRC: Button
02:09:39 MQT: stat/sonoff2/RESULT = {"POWER":"ON"}
02:09:39 MQT: stat/sonoff2/POWER = ON
02:09:40 CFG: Saved to flash at F8, Count 52, Bytes 3584`

at 02:09:39 there's a random Button1 press which i didn't trigger
i searched the closed issue, it happen with Shelly1 and i assume it's been fixed
however i'm using Slampher (v2) and the latest release 6.4.1 and the problem still there

please advise
Thank You

ascillato2, 2019-01-05:
Hi

Thanks for completing the Template. Very appreciated. 

The shelly issue is another thing. It has different hardware than slampher. It uses mains ac for switches and slampher don't. They have different switch management code for them.

* Did you connect anything to the button? Like another switch? 

* Did you modify the hardware in any way? 

* Did you connect any extra sensor?

thematica, 2019-01-05:
hi,
Thank you for replying

Did you connect anything to the button? Like another switch?
no, it only connect to the power source

Did you modify the hardware in any way?
no, only open the case, flash it and then close it again

Did you connect any extra sensor?
if you mean hardware, no

i connect it to Home Assistant and Eclipse MQTT, but i don't think they "touch" the button
i flashed 2 Slampher and both experience the same symptoms

currently i reflash the 1st to original FW just to make sure the hardware is good 
and yes, the original FW didn't randomly powered ON

i still have the 2nd on Tasmota if you need anything from it

Regards

ascillato2, 2019-01-05:
Please, try to erase all flash with esptool.py as explained in the wiki and flash the latest version from the development branch 6.4.1.5 from phttp://thehackbox.org/tasmota/

thematica, 2019-01-05:
i flashed the development version 6.4.1.5
do i need to create a log or we're just see if the problem still exist?

ascillato, 2019-01-05:
Yes, there were several fixes.

So, if your issue is solved, please close it. Thanks

thematica, 2019-01-05:
alright thanks

closed 2019-01-05 by thematica

tlc76, 2020-03-14:
Hello,

I have a similar issue but with some rather curious symptoms.

**Device:** Sonoff Slampher RF / Wifi
**Symptoms:** precisely at every 35-37 seconds, the device turns itself on/off.

All my Tasmotas are connected via a Mosquitto MQTT broker instance.

I also have a Sonoff RF Bridge with Tasmota and I can certainly say there is no action on the RF side (no "weird" 433MHz signal caught by the RF bridge).

weblog 4 on the Slampher console shows the following details:
> 22:38:31 WIF: Checking connection...
> 22:38:31 WIF: Connected
> 22:38:50 APP: Button1 multi-press 1
> 22:38:51 SRC: Button
> 22:38:51 MQT: tasmota/sonoff-slampher-1/stat/RESULT = {"POWER":"ON"}
> 22:38:51 MQT: tasmota/sonoff-slampher-1/stat/POWER = ON
> 22:38:51 CFG: Saved to flash at F7, Count 157, Bytes 4096
> 22:38:51 WIF: Checking connection...
> 22:38:51 WIF: Connected
> 22:39:11 WIF: Checking connection...
> 22:39:11 WIF: Connected
> 22:39:26 APP: Button1 multi-press 1
> 22:39:27 SRC: Button
> 22:39:27 MQT: tasmota/sonoff-slampher-1/stat/RESULT = {"POWER":"OFF"}
> 22:39:27 MQT: tasmota/sonoff-slampher-1/stat/POWER = OFF
> 22:39:27 CFG: Saved to flash at F6, Count 158, Bytes 4096
> 22:39:31 WIF: Checking connection...
> 22:39:31 WIF: Connected
> 22:39:51 WIF: Checking connection...
> 22:39:51 WIF: Connected
> 22:40:03 APP: Button1 multi-press 1
> 22:40:04 SRC: Button
> 22:40:04 MQT: tasmota/sonoff-slampher-1/stat/RESULT = {"POWER":"ON"}
> 22:40:04 MQT: tasmota/sonoff-slampher-1/stat/POWER = ON
> 22:40:04 CFG: Saved to flash at F5, Count 159, Bytes 4096
> 22:40:11 WIF: Checking connection...
> 22:40:11 WIF: Connected
> 22:40:16 APP: Button1 multi-press 1
> 22:40:17 SRC: Button
> 22:40:17 MQT: tasmota/sonoff-slampher-1/stat/RESULT = {"POWER":"OFF"}
> 22:40:17 MQT: tasmota/sonoff-slampher-1/stat/POWER = OFF
> 22:40:17 CFG: Saved to flash at F4, Count 160, Bytes 4096
> 22:40:31 WIF: Checking connection...
> 22:40:31 WIF: Connected
> 22:40:39 APP: Button1 multi-press 1
> 22:40:40 SRC: Button
> 22:40:40 MQT: tasmota/sonoff-slampher-1/stat/RESULT = {"POWER":"ON"}
> 22:40:40 MQT: tasmota/sonoff-slampher-1/stat/POWER = ON
> 22:40:40 CFG: Saved to flash at FB, Count 161, Bytes 4096
> 22:40:51 WIF: Checking connection...
> 22:40:51 WIF: Connected
> 22:41:12 WIF: Checking connection...
> 22:41:12 WIF: Connected
> 22:41:16 APP: Button1 multi-press 1
> 22:41:17 SRC: Button
> 22:41:17 MQT: tasmota/sonoff-slampher-1/stat/RESULT = {"POWER":"OFF"}
> 22:41:17 MQT: tasmota/sonoff-slampher-1/stat/POWER = OFF
> 22:41:17 CFG: Saved to flash at FA, Count 162, Bytes 4096
> 22:41:32 WIF: Checking connection...
> 22:41:32 WIF: Connected
> 22:41:52 WIF: Checking connection...
> 22:41:52 WIF: Connected
> 22:41:52 APP: Button1 multi-press 1
> 22:41:53 SRC: Button
> 22:41:53 MQT: tasmota/sonoff-slampher-1/stat/RESULT = {"POWER":"ON"}
> 22:41:53 MQT: tasmota/sonoff-slampher-1/stat/POWER = ON
> 22:41:53 CFG: Saved to flash at F9, Count 163, Bytes 4096

Although I have done the initial fw update without running the esptool erase command, I have done this at a later stage (esptool erase then upload Tasmota fw via an USB-to-TTL interface).
The problem persists.

As you can see in the above console log, the device seems to be actioned by the means of the physical button. Which is certainly false (no one is touching the button).

Has anyone else encountered this problem?
Is there any solution/workaround I could implement? Such as disabling the physical button?

Thank you,
Cristian

tlc76, 2020-03-14:
Applied this workaround and it seems everything is fine!
5 minutes and no more self power ons/offs.

Will keep an eye on it and let you know if issue still persists.

Thanks!

> If we can't solve this, may be it has some hardware defect that makes the switch to have noise. In that case, you can disable the button using generic module and configuring gpios as:
> 
> ```
>  RELAY1 in GPIO12 (Red Led and Relay) (0 = Off, 1 = On)
>  LED1i in GPIO13 Blue Led (0 = On, 1 = Off)
> ```