arendst/Tasmota #4825: Implement RDM6300 RFID reader

GianCann, 2019-01-05:
**Have you look for this feature in other issues and in the wiki?**
Yes.
There is only this article but not is usefull in real scenario because the TAG ID is read continuosly while is near the reader with generation of a lot of  identically  repetead MQTT messages
https://github.com/arendst/Sonoff-Tasmota/wiki/Wemos-D1-Mini-and-RDM6300

There is also an old request without any response:
https://github.com/arendst/Sonoff-Tasmota/issues/1599

**Is your feature request related to a problem? Please describe.**
It's an enhancement for using the low cost RDM6300 RFID reader to implement a door/access control entry based on low cost Tag ID.
Important: This reader need only one GPIO pin (rx serial) to work.

**Describe the solution you'd like**
The best (probabilly) solution is to implement this library:
https://github.com/arduino12/rdm6300
(Work also with ESP8266 chip)
 _
> * Connect the rdm6300 to VCC=5V, GND=GND, TX=any GPIO
>  * **Note: for ESP* or other 3.3V based microcontroller**,
>  *       you must divide the TX out of the rdm6300 to the 3.3V levels,
>  *       I used the following resistor divider:
>  *       GND--<2K resistor>--GPIO_2--<1K resistor>--TX(rdm6300).
_

The library read *only one time* the same ID TAG positioned near the reader until are not get away

**Describe alternatives you've considered**
Buy an PN532 reader...
https://github.com/arendst/Sonoff-Tasmota/issues/2915

**Additional context**
The reader on Amazon:
https://www.amazon.com/s/ref=nb_sb_noss?field-keywords=rdm6300

**(Please, remember to close the issue when the problem has been addressed)**

ozett, 2019-01-05:
i dont have such a reader, i have the pn532,
but i personally like getting all read-ins via MQTT to my node-red.
i process the debouncing in my node-red.

if anybody implements the requested debouncing, 
maybe he implements this as an option ?
like: 
[continious reading of tags] vs [single-read-in of same tag] ?

GianCann, 2019-01-05:
Hi ozett,
Why sending same MQTT message over the network, in a loop, when the data is the same?

ascillato2, 2019-01-05:
If you connect it directly to the esp8266 RX pin. You should see the commands coming from your tag reader, right?

In that case there is nothing extra to add. You can make the rules you want with the commands coming from your reader.

andrethomas, 2019-01-05:
@GianCann 
> Why sending same MQTT message over the network, in a loop, when the data is the same?

The current implementation of the PN532 driver scans for a tag every 250ms - If a tag is found it will suspend scanning for 2 seconds - If the same card is present again (or still present) then it will generate a new mqtt message and event.

This is done for two reasons

1) A normal person would not hold the card in place for more than 2 seconds.
2) For debugging purposes, we currently need it to repeat.

ozett, 2019-01-05:
> Why sending same MQTT message over the network, in a loop, when the data is the same?

i would send the TAG UID as long as the card is hold before the reader,
and do the supression (if a supression is wanted) anywhere else.
so not to change the default read-in behaviour.

that way, for shorthand examples, 
i could destingiuish long-hold vs. short-hold,
i could store holding-times of specific tags in database
i could program a logic for the relation of holding-time and my movement-dectector.
all not possible at all, when supression is made directly in the driver.

overall, a short debounce is normal on a switch, and so i guess on the reader,
a longer supression of multiple-read-ins obfuscates the plain function and maybe should better be done elsewhere (rules etc..) ?

gemu2015, 2019-01-10:
attached is a ready to use RDM6300 driver. just copy it into the sonoff folder. it claims the serial port, so nothing to set up except

#define USE_RDM6300

[xsns_41_rdm6300.ino.zip](https://github.com/arendst/Sonoff-Tasmota/files/2746041/xsns_41_rdm6300.ino.zip)

ascillato, 2019-01-10:
@gemu2015

Please, make a PR for Theo to review it. Thanks