arendst/Tasmota #4826: Feature request: parameter to set value for Dimmer "ON" command

sgofferj, 2019-01-05:
Currently, Tasmota reacts to cmnd/dimmer "ON" with setting the dimmer to 1%. Different smart home control systems, among others, OpenHAB, assume "ON" to mean 100%.

I'd like to suggest a SetOption parameter for dimmable devices which would define the desired dimmer value in % to be set when the devices receives a dimmer "ON" command. Alternatively, the SetOption value could be 0/1 for 1%/100%.

Using the "ON" command for a dimmer is not always avoidable. E.g. Alexa sends "ON" instead of a % value when commanded "Alexa, turn on <device>", even if the device is a dimmer.

ascillato, 2019-01-05:
Hi,

For On you have power on. 

Alexa command are interpreted by the hue bridge emulation and executed the proper command. 

Please, use the configuration examples of the wiki. Thanks

sgofferj, 2019-01-05:
In OpenHAB, Alexa commands are interpreted by the Alexa binding.

ascillato, 2019-01-05:
Ok. Good to know.

So, you can use Alexa directly to Tasmota or with Openhab in between. 

Using Alexa directly to Tasmota is working fine and no need to add anything extra to dimmer or power on

ascillato2, 2019-01-05:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-01-05 by ascillato2

sgofferj, 2019-01-05:
No, this issue has NOT been answered!
The offered solution is invalid as it requires to configure an additional control way.
The point of a central control system like OpenHAB is to NOT have to configure different devices to different services. Having to configure an additional control path for Tasmota devices is counter-intuitive to a point where it becomes absurd.
@ascillato: You obviously do not understand home-automation.