arendst/Tasmota #4827: Shelly1 use button instead of switch

newmindJP, 2019-01-05:
I flashed my shelly1 with tasmota v6.4.1 from the provided bin release and used available module config "Shelly 1"
I'm playing around since several days now but without success. I connected a push button instead a switch to the input (GPIO5) - I want following functionality:
- single press -> toggle output
- double press -> mqtt msg
- hold -> mqtt msg

this should be supported as described at wiki.
I used following config:
SwitchMode1 6
SetOption1 1
SetOption11 1
ButtonTopic shelly1/1/button

single press still works, but double and hold do not generate a message
if I change ButtonTopic  to SwitchTopic, then I get a mqtt message at single press and hold, but no message at double and no output toggle any longer.
I also tried to used rules as described in the wiki without success - seems that switch does not support double-press.

Today I started from scratch and used instead of "shelly 1" module config the "generic" module.
Following settings
Module 18
GPIO4 21
GPIO5 90
SwitchMode1 0
SetOption1 1
SetOption11 1
ButtonTopic shelly1/1/button

with this setting almost everything works, beside the fact, that the push-button has to be inverted. I cannot find any settings to invert the push-button functionality like with "SwitchMode1 1" - seems there is no "ButtonMode1"

Any idea how I can fix this?

ascillato2, 2019-01-05:
Hi

Buttontopic is for buttons. Switchtopic is for switches. Those depends on how you configure your gpios on module config or how is the template defined for each device. In the case of shelly 1 it is defined as a switch with no pull ups. 

The gpio definition as button is not supported in shelly. But you can use a switch with the correct switchmode to use your pushbutton. 

Anyway, shelly uses a special code for managing the gpios because of the hardware, so, with the generic approach you could experience ghost switching. You need to use the standard shelly 1 selection in module config and then set the switchmode and switchtopic you want.

ascillato2, 2019-01-05:
If you need further assistant please find us in the Tasmota Chat. Thanks

closed 2019-01-05 by ascillato2

twieland, 2019-04-24:
Hi,

I actually have the same question: is it possible to use a Shelly1 with three different actions for single/double/hold press? Like here https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#16-using-an-external-button-with-single-press---double-press-and-hold

If I understand correctly it's not possible because the shelly input only works as switch?