arendst/Tasmota #4828: Kingart Wall Dimming module 8285

atideath, 2019-01-05:
Hello.

I ordered one new Dimmer Wall switch from Aliexpress. It is working correctly with ewelink but I don't like it. I would like to use with Tasmota FW. Anyone can help me how to flash this module? Where I find the pinouts and It will work with tasmota? I attach picture from this module.

This is the link from this product: [DHLGate](https://www.dhgate.com/product/wireless-home-automation-ewelink-app-control/429808787.html)

[image]

ascillato2, 2019-01-05:
### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-01-05 by ascillato2

Patrickl3103, 2019-02-20:
Hallo. Ich habe das gleiche Problem. Flashen hab ich geschafft, aber welche Einstellungen muss ich jetzt vornehmen in Tasmota? Finde keine Geräteeinstellung um den Dimmer richtig nutzen zu können.
Hast du da was rausbekommen?

hawkjohn, 2019-03-18:
In order to get answers to upper questions, I'd like to push this thread. Has anyone already made a support request? I have the same issue with this Dimmer and I have no idea how to connect it to my smart home via tasmota. I have already flashed the Dimmer with:
Program Version 6.4.1.22(99d20f8-sonoff)
Build Date & Time 2019-03-16T18:02:00
Core/SDK Version 2_5_0/3.0.0-dev(c0f7b44)
 Can anyone help please, I'm still a noob?