arendst/Tasmota #4831: Setoption19 ON - make my Tasmota wachdog reset

peiperakos, 2019-01-05:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Development/Compiler/Upload tools used : Arduino IDE
- [x] Hardware used : NodeMCU v1
- [ ] If a pre-compiled release or development binary was used, which one? :
- [x] You have tried latest release or development binaries? :
- [x] Provide the output of command ``status 0`` :
```
20:43:26 RSL: STATUS = {"Status":{"Module":18,"FriendlyName":["r00t Home","r00t Home2","r00t Home3","r00t Home4"],"Topic":"r00t","ButtonTopic":"0","Power":16,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
20:43:26 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software Watchdog","Uptime":"0T00:21:37","StartupUTC":"2019-01-05T18:21:49","Sleep":50,"BootCount":40,"SaveCount":229,"SaveAddress":"3FA000"}}
20:43:26 RSL: STATUS2 = {"StatusFWR":{"Version":"6.4.1.4(sonoff)","BuildDateTime":"2019-01-05T19:06:44","Boot":31,"Core":"2_5_0_BETA2","SDK":"3.0.0-dev(c0f7b44)"}}
20:43:26 RSL: STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["r00t",""],"TelePeriod":300,"SetOption":["00028001","558180C0","00000200"]}}
20:43:26 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":777,"Free":3316,"Heap":18,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640C8","FlashMode":2,"Features":["00000809","4FFAE79C","000387A0","23B617CE","00003BC0"]}}
20:43:26 RSL: STATUS5 = {"StatusNET":{"Hostname":"r00t-0000","IPAddress":"192.168.1.102","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"60:01:94:66:E6:85","Webserver":2,"WifiConfig":1}}
20:43:26 RSL: STATUS7 = {"StatusTIM":{"UTC":"Sat Jan 05 18:43:26 2019","Local":"Sat Jan 05 20:43:26 2019","StartDST":"Sun Mar 31 03:00:00 2019","EndDST":"Sun Oct 27 04:00:00 2019","Timezone":99,"Sunrise":"07:46","Sunset":"17:07"}}
20:43:26 RSL: STATUS10 = {"StatusSNS":{"Time":"2019-01-05T20:43:26","Switch5":"OFF","AM2301":{"Temperature":22.7,"Humidity":45.1},"BMP180":{"Temperature":21.2,"Pressure":1012.3},"PressureUnit":"hPa","TempUnit":"C"}}
20:43:26 RSL: STATUS11 = {"StatusSTS":{"Time":"2019-01-05T20:43:26","Uptime":"0T00:21:37","Vcc":3.025,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"ON","Wifi":{"AP":1,"SSId":"r00t","BSSId":"78:96:82:32:BE:D6","Channel":11,"RSSI":48}}}
```

Hello Friends! i am new ! 
if i run "setoption19 1" my Tasmota wachdog reset!
as you can see from my status i have on i2c bmp180 and one DHT22
if in hardware setting disable the sda,scl,and dht pin the command "setoption19 1" execute perfect without a problem!
****if in hardware setting enable the sda,scl and dht22 pin the command "setoption19 1" wachdog reset!!
i try with one of the two sensor and again reset!
if i enable in generic module  pin like sda or scl or dht22 the  "setoption19 1" wachdog reset!!

ascillato2, 2019-01-05:
I can reproduce your issue in core 2.5.0 and also in 2.4.2. Investigating. Thanks for Reporting.

peiperakos, 2019-01-06:
to be good for the important program that is being maintained! Very Good Job for Tasmota!

peiperakos, 2019-01-08:
Sorry arendst  i cant unterstant where i have to add this yield in witch loop of tasmota ?
and what is yield?
sorry i am new with this !

ps i dont have home assistant server on run i compile this command (setoption19 ) only for test and i realrealise that the tasmota reset!

peiperakos, 2019-01-08:
ohh now unterstand wait i download 6.4.1.8

peiperakos, 2019-01-08:
i download and compile 6.4.1.8 and again {"RestartReason":"Software Watchdog"}

peiperakos, 2019-01-08:
ok now i run the command setoption19 1 for 3 times and the 3-4 times the command complile perfect!
and after to close thies setoption19 0 for 3-4 times after restart compile again perfect but not with the first time as you can say !!!

thenx for your suppurt!

andrethomas2, 2019-01-08:
Closing this issue as you seem to have solved it - If it re-occurs please do not hesitate to make contact through the channels provided below.

Thanks for the information and last update, much appreciated.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-01-08 by andrethomas2