arendst/Tasmota #4835: MPU6050 and DS18B20 not working together

erwini, 2019-01-05:
**Describe the bug**
DS18B20 not seen, if compiling with option USE_MPU6050 & USE_MPU6050_DMP. Option USE_DS18x20 always enabled

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [X] _Searched the problem in issues and in the wiki_
- [X] _Hardware used_ : Wemos D1 mini + MPU-6050 (sda/scl on gpio 4&5), DS18B20 on gpio 0
- [X] _Development/Compiler/Upload tools used_ : Arduino IDE 1.8.8 + esptool.py
- [X] _If a pre-compiled release or development binary was used, which one?_ : self compiled from sources of 6.4.1
- [X] _You have tried latest release or development binaries?_ : no, no mpu6050 support.
- [X] _Provide the output of command_``status 0`` :
```
22:20:45 CMD: status 0
22:20:45 MQT: stat/sonoff/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff","Sonoff2"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
22:20:45 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"External System","Uptime":"0T00:01:35","StartupUTC":"2019-01-05T21:19:10","Sleep":200,"BootCount":34,"SaveCount":59,"SaveAddress":"F9000"}}
22:20:45 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.4.1(sonoff)","BuildDateTime":"2019.01.05 15:53:07","Boot":5,"Core":"2_5_0_BETA2","SDK":"3.0.0-dev(c0f7b44)"}}
22:20:45 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["xxxxxx","xxxxxxxx2"],"TelePeriod":300,"SetOption":["00008009","5581C0C0","00000000"]}}
22:20:45 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":711,"Free":292,"Heap":17,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000407","0FDAE794","000383A1","23B617CE","00003BC8"]}}
22:20:45 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-2041","IPAddress":"10.184.0.186","Gateway":"10.184.0.64","Subnetmask":"255.255.255.0","DNSServer":"10.184.0.168","Mac":"84:F3:EB:B3:A7:A9","Webserver":2,"WifiConfig":4}}
22:20:45 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"10.184.0.171","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_B3A7A9A9","MqttUser":"xxmqtt","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
22:20:45 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Sat Jan 05 21:20:45 2019","Local":"Sat Jan 05 22:20:45 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:15","Sunset":"16:39"}}
22:20:45 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-01-05T22:20:45","MPU6050":{"Temperature":22.5,"AccelXAxis":-555.00,"AccelYAxis":26.00,"AccelZAxis":2326.00,"GyroXAxis":64.00,"GyroYAxis":-5.00,"GyroZAxis":-12.00},"TempUnit":"C"}}
22:20:45 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-01-05T22:20:45","Uptime":"0T00:01:35","Vcc":2.997,"SleepMode":"Dynamic","Sleep":200,"LoadAvg":4,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":2,"SSId":"xxxxxxxx2","BSSId":"AC:C7:A6:22:8E:EE","Channel":11,"RSSI":74}}}

```

**To Reproduce**
Sensor DS18B20 is always connected.
If using binary compiled without MPU6050 support, Temperature from DS18B20 is shown in the webui/console.
If using binary compiled with MPU6050 support, Temperature from DS18B20 is not shown in the webui/console.
Sensor BME280 (I2C Temp/Hum/Pres) works always (if connected to I2C). No difference to above issue, if BME280 sensor is connected to I2C or not.

**Expected behavior**
DS18B20 and MPU6050 are working at the same time.

**Screenshots**
No Screenshots

**Additional context**
Expected behavior by design or bug?

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2019-01-05:
Hi,

Please try today's version from development branch (6.4.1.6). Some changes regarding I2C were made.

Thanks

erwini, 2019-01-05:
Will try it, but DS18B20 is **not** using I2C, it is using 1wire ...

erwini, 2019-01-06:
Tested 6.4.1.6

- without option mpu6050 enabled: no working DS18B20
- with option mpu6050 enabled: no working DS18B20
- disabled user_config_override: no working DS18B20

again tested 6.4.1 binary (self compiled)

- without option mpu6050 enabled: working DS18B20

So additional it seems in 6.4.1.6 support of DS18B20 is complete gone.
Checked also, if configuration for DS18x20 is always present in all steps. That was the case.

Einstein67, 2019-01-06:
> So additional it seems in 6.4.1.6 support of DS18B20 is complete gone.

DS18x20 is working fine as single- (DS18B20_INTERNAL_PULLUP) and as multi-sensor (DS18x20_LEGACY) with 6.4.1.6

erwini, 2019-01-06:
Thanks for hint.
6.4.1.6:
- DS18x20_LEGACY is working here as well
- DS18x20 is not working
- DS18x20_LEGACY and MPU6050* is working together.

So it seems DS18x20 is broken. If i go with DS18x20_LEGACY, i have to rewrite my automation, because MQTT messages are different.
And DS18x20 is the default used in the sources.

arendst, 2019-01-06:
@erwini did you enable define W1_PARASITE_POWER?

erwini, 2019-01-06:
**@Arendst** No. DS18B20 is connected direct to 3v3 and Ground, Data to D3

Staars, 2019-01-07:
Good morning,

I wired something together and got this:
`10:02:05 MQT: tele/sonoff/SENSOR = {"Time":"2019-01-07T10:02:05","DS18B20-1":{"Id":"0417819957FF","Temperature":21.9},"DS18B20-2":{"Id":"041781AFE6FF","Temperature":21.6},"MPU6050":{"Temperature":22.7,"AccelXAxis":52.00,"AccelYAxis":-676.00,"AccelZAxis":450.00,"GyroXAxis":45.00,"GyroYAxis":-45.00,"GyroZAxis":39.00},"TempUnit":"C"}`

I know that it is easy to overlook something and I am still in the process of triple-checking my compile settings, just to be sure, that this a correct report.
Basically this is 6.4.1.7 compiled with 2.4.2 and only the two sensors selected plus some inserted debug code. I am still very unsure, if this is a coincidence (maybe by adding some delay here and there) but I will continue digging.
Another question to @erwini : Did you connect the DS18b20 with additional pull ups between data and vcc?

Staars, 2019-01-07:
Some update:
I did further tests and started "as-from-scratch-as-possible" (flash erase, local git erase).
Then I only uncommented `#define USE_MPU6050   ` in sonoff_post.h and built the "sonoff-sensors"-target.
It worked OOB and the only thing I noticed is, that sometimes only one sensors shows up after restart.

`11:40:05 MQT: stat/sonoff/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
11:40:05 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:05:25","StartupUTC":"2019-01-07T10:34:40","Sleep":50,"BootCount":25,"SaveCount":35,"SaveAddress":"F9000"}}
11:40:05 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.4.1.7(sensors)","BuildDateTime":"2019-01-07T11:25:51","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
11:40:05 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":[xxxxxx",""],"TelePeriod":300,"SetOption":["00008009","5581C0C0","00000000"]}}
11:40:05 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":552,"Free":4191636,"Heap":23,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640E0","FlashMode":3,"Features":["00000809","0FDEE794","2403A3A4","B7FFBFCC","000ABBC8"]}}
11:40:05 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-7412","IPAddress":"192.168.0.112","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"xxxxxxxxxx","Webserver":2,"WifiConfig":4}}
11:40:05 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"xxxxxxxxx","MqttPort":xxxx,ClientMask":"DVES_%06X","MqttClient":"DVES_169CF4","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
11:40:05 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Mon Jan 07 10:40:05 2019","Local":"Mon Jan 07 11:40:05 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:42","Sunset":"17:10"}}
11:40:05 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-01-07T11:40:05","ANALOG":{"A0":1},"DS18B20-1":{"Id":"0417819957FF","Temperature":22.2},"DS18B20-2":{"Id":"041781AFE6FF","Temperature":21.9},"MPU6050":{"Temperature":23.9,"AccelXAxis":1218.00,"AccelYAxis":-14834.00,"AccelZAxis":10162.00,"GyroXAxis":811.00,"GyroYAxis":532.00,"GyroZAxis":-294.00},"TempUnit":"C"}}
11:40:05 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-01-07T11:40:05","Uptime":"0T00:05:25","SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"Wifi":{"AP":1,"SSId":"xxxxxxxx","BSSId":"xxxxxxxxxxxx","Channel":1,"RSSI":46}}}
`

So at the moment I can not really reproduce the bug in my setup (besides the restart issue).

erwini, 2019-01-07:
**Question** @erwini : Did you connect the DS18B20 with additional pull ups between data and vcc?
**Answer** No pull ups. It works without pullups with 6.4.1, changing to 6.4.1.6 DS18B20 does not work.

@Staars:
At the moment i use only one DS18B20, but want to use more.

I compiled this way (working):

File *user_config_override.h*

```
// -- One wire sensors ----------------------------
#undef USE_DS18x20
                                                 // WARNING: Select none for default one DS18B20 sensor or enable one of the following two options for multiple sensors
#define USE_DS18x20_LEGACY                       // Optional for more than one DS18x20 sensors with dynamic scan using library OneWire (+1k5 code)
//#define USE_DS18x20                              // Optional for more than one DS18x20 sensors with id sort, single scan and read retry (+1k3 code)
//  #define W1_PARASITE_POWER                      // If using USE_DS18x20 then optimize for parasite powered sensors
//  #define DS18B20_INTERNAL_PULLUP                 // Use INPUT_PULLUP internal pullup resistors for single DS18B20

// -- I2C Sensor ------------------------------
  #define USE_MPU6050                            // Enable MPU6050 sensor (I2C address 0x68 AD0 low or 0x69 AD0 high) (+3K3 of code and 188 Bytes of RAM)
    #define USE_MPU6050_DMP                      // Enable in MPU6050 to use the DMP on the chip, should create better results (+8k6 of code)
```

I compiled this way (not working):

File *user_config_override.h*

```
// -- One wire sensors ----------------------------
#undef USE_DS18x20
                                                 // WARNING: Select none for default one DS18B20 sensor or enable one of the following two options for multiple sensors
//#define USE_DS18x20_LEGACY                       // Optional for more than one DS18x20 sensors with dynamic scan using library OneWire (+1k5 code)
#define USE_DS18x20                              // Optional for more than one DS18x20 sensors with id sort, single scan and read retry (+1k3 code)
//  #define W1_PARASITE_POWER                      // If using USE_DS18x20 then optimize for parasite powered sensors
//  #define DS18B20_INTERNAL_PULLUP                 // Use INPUT_PULLUP internal pullup resistors for single DS18B20

// -- I2C Sensor ------------------------------
  #define USE_MPU6050                            // Enable MPU6050 sensor (I2C address 0x68 AD0 low or 0x69 AD0 high) (+3K3 of code and 188 Bytes of RAM)
    #define USE_MPU6050_DMP                      // Enable in MPU6050 to use the DMP on the chip, should create better results (+8k6 of code)
```

- Changes done only in `user_config_override.h` , enabling it in my_user_config.h.
- No changes in sonoff_post.h and built-target. So maybe the built-target does not match here?!

erwini, 2019-01-07:
Forgoten, compiled with 2.5.0-beta2

arendst, 2019-01-07:
So the simple solution is that you need a pull up between data and Vcc.

That it worked with 6.4.1 is pure luck as it all depends on power load and noise.

erwini, 2019-01-07:
Tested with 4k7 pullup between Vcc 3v3 and data and 6.4.1.6 and USE_DS18x20: not working.
Compile options in `user_config_override.h`:

```
// -- One wire sensors ----------------------------
#undef USE_DS18x20
                                                 // WARNING: Select none for default one DS18B20 sensor or enable one of the following two options for multiple sensors
//#define USE_DS18x20_LEGACY                       // Optional for more than one DS18x20 sensors with dynamic scan using library OneWire (+1k5 code)
#define USE_DS18x20                              // Optional for more than one DS18x20 sensors with id sort, single scan and read retry (+1k3 code)
//  #define W1_PARASITE_POWER                      // If using USE_DS18x20 then optimize for parasite powered sensors
//  #define DS18B20_INTERNAL_PULLUP                 // Use INPUT_PULLUP internal pullup resistors for single DS18B20

// -- I2C Sensor ------------------------------
  #define USE_MPU6050                            // Enable MPU6050 sensor (I2C address 0x68 AD0 low or 0x69 AD0 high) (+3K3 of code and 188 Bytes of RAM)
    #define USE_MPU6050_DMP                      // Enable in MPU6050 to use the DMP on the chip, should create better results (+8k6 of code)
```

Why is it working with USE_DS18x20_LEGACY and without pullup in the tests before? Correct noise :-) ?

Einstein67, 2019-01-07:
> Why is it working with USE_DS18x20_LEGACY and without pullup in the tests before

Some Wemos-GPIOs have pullups on board (D3). 

Have tested all DS18x20 drivers and all are working perfect for me ....

erwini, 2019-01-07:
Using D3 as well ....
Exchanged DS18B20 with another one. No luck.
Running out of idears. 
Some Debug/Print Commands possible in the sourcecode ? Never done this in this enviroment.
Also checked the binary with `hexdump -C | grep 18x20` looking for traces of the included drivers for DS18x20. Found. So its included.

erwini, 2019-01-09:
More tests with DS18B20.
Using D2 with pullup. Not working
Used another Wemos D1 mini. Same behavior.
Used a plain 8266 modul. Same behavior.
Works with 6.4.1 but not with 6.4.1.6 (selfcompiled). It looks like something is wrong with the compiler.

erwini, 2019-01-13:
Hi,
@ascillato 
Problem with the compiler or some settings in some \*config\*.h file is only a suspicion. I have no errormessages. I am a little bit busy next time, so it takes a while now. Maybe i wait for the next public release, use the the precompiled .bin files, test the DS18B20 support with the .bin file and then i will starting selfcompiling.

ascillato2, 2019-01-13:
Ok. So if you will wait for next release, I will put this issue on Hold. Then, when you have time to test, please ask here and we can reopen it of needed. Thanks for all testings.

closed 2019-01-13 by ascillato2

teddybee, 2020-06-17:
I have similar problems then the op. The Wemos D1 mini wont boot if there is a 4.7K or 2.2K pullup with DS18b20 and AM2301.
I connected the data of AM2301 to D4 and the dallas to D8. If I wired the AM2301 without pull-up, the Wemos boots, but after 1-2 hours it don't read sensor values. 
Unfortunately I have never seen the two sensors working together at the same time...
One more thing, I use the board with 18650 battery, at the moment it has aroucd 4V charge.