arendst/Tasmota #4836: 6.4.x - MQTT Full Topic field in Web Interface - can't change value

ketchup007, 2019-01-05:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

**Describe the bug**
In MQTT section of configuration Web interface I can't change value of FULL TOPIC field. After wrote a new value, back old (%topic%/%prefix%/) value after automatic restart.

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [X] _Searched the problem in issues and in the wiki_
- [X] _Hardware used_ : sonoff th10, th16, pow2, lolin,... 
- [X] _Development/Compiler/Upload tools used_ : uploaded by web
- [X] _If a pre-compiled release or development binary was used, which one?_ : 6.4, 6.4.1
- [X] _You have tried latest re lease or development binaries?_ : http://thehackbox.org/tasmota/archive/20190105-200225-9da900d-sonoff.bin
- [X] _Provide the output of command_``status 0`` :
```
STATUS 0 OUTPUT HERE - DO NOT DELETE THE MARKERS ABOVE AND BELOW THIS LINE
```
00:23:14 CMD: status 0
00:23:14 MQT: TH16_002/stat/STATUS = {"Status":{"Module":4,"FriendlyName":["TH16_002"],"Topic":"TH16_002","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:23:14 MQT: TH16_002/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-PL.bin","RestartReason":"Software/System restart","Uptime":"0T00:05:03","StartupUTC":"2019-01-05T23:18:11","Sleep":50,"BootCount":81,"SaveCount":256,"SaveAddress":"F7000"}}
00:23:14 MQT: TH16_002/stat/STATUS2 = {"StatusFWR":{"Version":"6.4.1.6(9da900d-sonoff)","BuildDateTime":"2019-01-05T20:01:55","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
00:23:14 MQT: TH16_002/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["service",""],"TelePeriod":300,"SetOption":["000A8009","558180C0","00000200"]}}
00:23:14 MQT: TH16_002/stat/STATUS4 = {"StatusMEM":{"ProgramSize":522,"Free":480,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440E0","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
00:23:14 MQT: TH16_002/stat/STATUS5 = {"StatusNET":{"Hostname":"TH16_002","IPAddress":"10.3.250.241","Gateway":"10.3.250.1","Subnetmask":"255.255.255.0","DNSServer":"10.3.250.1","Mac":"2C:3A:E8:07:3F:3B","Webserver":2,"WifiConfig":5}}
00:23:14 MQT: TH16_002/stat/STATUS6 = {"StatusMQT":{"MqttHost":"10.3.255.197","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_073F3B","MqttUser":"test","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
00:23:14 MQT: TH16_002/stat/STATUS7 = {"StatusTIM":{"UTC":"Sat Jan 05 23:23:14 2019","Local":"Sun Jan 06 00:23:14 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:43","Sunset":"17:09"}}
00:23:14 MQT: TH16_002/stat/STATUS10 = {"StatusSNS":{"Time":"2019-01-06T00:23:14","AM2301":{"Temperature":19.3,"Humidity":50.3},"TempUnit":"C"}}
00:23:14 MQT: TH16_002/stat/STATUS11 = {"StatusSTS":{"Time":"2019-01-06T00:23:14","Uptime":"0T00:05:03","Vcc":3.498,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"service","BSSId":"D6:CA:6D:08:80:56","Channel":3,"RSSI":54}}}

**To Reproduce**
_Steps to reproduce the behavior:_

**Expected behavior**
_A clear and concise description of what you expected to happen._

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Additional context**
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2019-01-05:
Please, share the fulltopic you want to add

ketchup007, 2019-01-05:
_%prefix%/%topic%/_ like default in (), but always is set _%topic%/%prefix%/_

ascillato, 2019-01-05:
please, try `setoption19 1`

ascillato, 2019-01-06:
If the issue remains, seems that there is something wrong left in your flash. In that case, if you do a full flash erase with esptool.py (as explained in the wiki) and then flash again the precompiled bins you should solve your issue.

ketchup007, 2019-01-06:
can I full flash erease remotely? same of sonoffs are installed...

ascillato, 2019-01-06:
Fully can't be done by wifi. But you can try partially:

In the console type `reset 5`

The only data that will remain is wifi credentials for you to connect again

ascillato2, 2019-01-06:
If that doesn't work the last option is to uninstall your sonoff and do a full flash erase with esptool.py (as explained in the wiki) and then flash again the precompiled bins in order to solve your issue.

Please, tell us if that works for you. Thanks

closed 2019-01-06 by ascillato2

kxkayser, 2019-02-09:
I don’t have a lot of experience but the same thing happened to me. It happened when I enabled auto discovery for Home Assitant.  I assume this is because it publishes an mqtt message for HA in a specific format. Did you recently enable this feature?  When I restored a prior config without auto discovery, it worked fine. Then when I enabled discovery, the issue reappeared. I don’t know if it’s designed like this or is a bug.

sincze, 2019-05-17:
Indeed when you issue: "setoption19 0" from the console it is now possible to change full topic again to "tasmota/%prefix%/%topic%/"

llSimsll, 2019-09-06:
Can confirm "setoption19 0" is working for me - sonoff basic v.6.4.1
Can't say why this one sonoff has it set to on, all other devices is default off.