arendst/Tasmota #4839: Increase HX711 resolution

nicolasbisi, 2019-01-06:
**Have you look for this feature in other issues and in the wiki?**
Yes

**Is your feature request related to a problem? Please describe.**
Maybe, not sure. I'm only getting integer values.

**Describe the solution you'd like**
Maybe having a Sensor34 7 that switches between Kg and g. Or just add Kg with more resolution (x,xx instead of integer only)

Cheers

arendst, 2019-01-06:
Look in wiki for command weightres

nicolasbisi, 2019-01-06:
Ah. Nice. Thanks, @arendst.

closed 2019-01-06 by nicolasbisi

pico563, 2019-05-11:
@arendst 
thank you for all your great work
I spent a hole day trying to find how to change the resolution but unfortunately there is no Wiki for HX711.
I changed in the sketch to get it work as gram (i removed the / 1000)
finally I found this command **"weightres"**
I still don't understand **Item weight (kg)** is used for.
please add a wiki for HX711
thanks

meingraham, 2019-05-11:
@pico563 Arggh! WeightRes was missing from the wiki. That was the missing piece I was hoping for. I added the WeightRes command to the wiki.

The whole topic of how to best document supported sensors is being discussed. When this portion of the wiki is restructured, we will look at all the supported sensors, out of date sensors, future sensors, etc.

Thanks for being persistent.

Mike

pico563, 2019-05-11:
hi @meingraham 
I found it after our last night discussion (zee)
thanks

pico563, 2019-05-11:
@meingraham @ascillato2 
I still don't understand **_Item weight (kg)_** is used for.

pico563, 2019-05-11:
[https://dl3.pushbulletusercontent.com/rbr2oUH5jrydRe3uIiQXtwak1a9fxeyZ/Screenshot_20190511-150149.png](url)

pico563, 2019-05-16:
@meingraham
@ascillato2

about sensor34  6
it shows item weight in gram and not in decigram.
pls. check
```
`6`  = show single item weight in decigram
|`6 <value>`  = set single item weight in decigrams. Once the item weight is set, when items are added to the scale, the telemetry message will report the number of items.
```
```
23:00:01 CMD: Sensor34  6
23:00:01 RSL: RESULT = {"Sensor34":{"WeightRef":250,"WeightCal":261,"WeightMax":10000,"WeightItem":250.0}}

23:00:03 RSL: SENSOR = {"Time":"2019-05-16T23:00:03","HX711":{"Weight":1.534,"Count":6}}
```

meingraham, 2019-05-16:
It appears that you set the weight per item to be 250. Is the reported `Weight` in grams or Kilograms? If it's Kilograms, then I agree, the item weight setting is grams. Please confirm that the weight on your scale was 1.5 Kg.

pico563, 2019-05-16:
confirmed .

meingraham, 2019-05-16:
Wiki updated.

dexterbot80, 2024-08-29:
> Look in wiki for command weightres

Hello @arendst, hello @meingraham Where is located that wiki page?
I want to learn how to display the weight in grams.
Thank you

sfromis, 2024-08-29:
The unit is always kilograms, but using the command [WeightRes](https://tasmota.github.io/docs/Commands/#weightres), you can set a precision of 3 decimals for grams.

dexterbot80, 2024-08-29:
> The unit is always kilograms, but using the command [WeightRes](https://tasmota.github.io/docs/Commands/#weightres), you can set a precision of 3 decimals for grams.

Thank you @sfromis

sfromis, 2024-08-29:
In general, you can expect things working with Tasmota for ESP8266 to also work with ESP32, but with Mr. Murphy not being "very predictable", I'm nok making absolute claims :wink:

dexterbot80, 2024-08-29:
> In general, you can expect things working with Tasmota for ESP8266 to also work with ESP32, but with Mr. Murphy not being "very predictable", I'm nok making absolute claims 😉

Hello, It works with ESP32 but i have one Chinese weighing machine that have bluetooth connection with the screen.
They use an Chinese IC chip that have I2C connection to - 3.3V SDA SCL GND pins available on the board.
It will be great if we manage to connect TASMOTA to this and make-it MQTT compatible :)

sfromis, 2024-08-29:
You already have [another thread](https://github.com/arendst/Tasmota/discussions/22053) about that **unrelated** question.

dexterbot80, 2024-08-30:
> You already have [another thread](https://github.com/arendst/Tasmota/discussions/22053) about that **unrelated** question.

Yeah, Maybe we make-it .....:)
I connected the 400KG weight cell to the HX711 and encountered a rather interesting problem.
When I try to calibrate with a known weight, it displays a much higher weight.
At a weight of 291 grams it displays 311 - 312 after calibration and at my weight of 76200 grams after calibration it displays 78800-78900 grams.
I used two types of boards with HX711 powered by both 5V and 3.3V and they behave the same.
I use the latest version of tasmota sensors [from here](https://tasmota.github.io/install/) and the pins for DT and SCK are D4 and D5 at ESP32.
Has anyone encountered this problem?

dexterbot80, 2024-09-02:
Hello, @arendst, @meingraham 
Can someone tell me where I can find the full calibration process for the HX711? I kept trying but I can't calibrate the scale.

I designed this for Arduino and the calibration works, but I would like to be able to use Tasmota - programmers much more skilled than me worked here :)

`#include <Ethernet.h>
#include <PubSubClient.h>
#include <HX711.h>
#include <EEPROM.h>

// Ethernet settings
byte mac[] = { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED };
IPAddress ip(192, 168, 1, 21);  // IP address of Arduino
IPAddress server(192, 168, 1, 200);  // MQTT Server IP

// MQTT settings
const char* mqttUser = "openhabian";
const char* mqttPassword = "openhabian";
const char* mqttTopicWeight = "home/loadcell/weight";
const char* mqttTopicCalibration = "home/loadcell/calibration";

// Define Client ID
#define CLIENT_ID "Client_001"  // Modifică acest ID pentru fiecare dispozitiv

// HX711 load cell settings
#define DOUT  6
#define CLK  7
HX711 scale;

float calibration_factor = 1.0;  // Default calibration factor
float known_weight = 2000.0;     // Known weight placed on the load cell in grams

// EEPROM Addresses
#define EEPROM_CALIBRATION_FACTOR_ADDR 0

// Ethernet client
EthernetClient ethClient;

// MQTT client
PubSubClient client(ethClient);

// Define RESET_EEPROM macro
#define RESET_EEPROM 1  // Change this to 1 to reset EEPROM

// Function to write a float value to EEPROM
void EEPROM_writeFloat(int address, float value) {
  byte* data = (byte*)(void*)&value;
  for (int i = 0; i < sizeof(float); i++) {
    EEPROM.write(address + i, data[i]);
  }
}

// Function to read a float value from EEPROM
float EEPROM_readFloat(int address) {
  float value = 0.0;
  byte* data = (byte*)(void*)&value;
  for (int i = 0; i < sizeof(float); i++) {
    data[i] = EEPROM.read(address + i);
  }
  return value;
}

// Function to reset EEPROM
void resetEEPROM() {
  for (int i = 0; i < EEPROM.length(); i++) {
    EEPROM.write(i, 0xFF);  // Write 0xFF to every address in EEPROM
  }
  Serial.println("EEPROM reset completed.");
}

[... 141 lines cut ...]
  // Check if weight is less than 10 grams and reset scale if true
  if (weight < 10) {
    weight = 0; // Set weight to 0 grams
    scale.tare(); // Reset scale to zero
    Serial.println("Weight is less than 10 grams. Scale reset to zero.");
  }

  // Print the weight to Serial
  Serial.print("Weight: ");
  Serial.print(weight);
  Serial.println(" grams");

  // Convert the weight to string and publish it
  char weightString[8];
  if (weight < 10) {
    snprintf(weightString, sizeof(weightString), "0"); // Publish 0 grams if weight is less than 10 grams
  } else {
    dtostrf(weight, 1, 2, weightString);
  }
  client.publish(mqttTopicWeight, weightString);

  // Delay before next reading
  delay(2000);  // 2-second delay between readings
}
`

dexterbot80, 2024-09-03:
> Look in wiki for command weightres

Hello @arendst 
You kill me with this phrase :)
Where is the wiki for HX711 located?
I searched [here](https://tasmota.github.io/docs/Supported-Peripherals/) but the HX711 does not have a dedicated page

Thank you

barbudor, 2024-09-03:
All commands are on the Commands page
https://tasmota.github.io/docs/Commands/

dexterbot80, 2024-09-03:
> All commands are on the Commands page https://tasmota.github.io/docs/Commands/

Thank you @barbudor 
Now I found for Sensor34
I hope I can calibrate :)
Thank again 1000 times :)

dexterbot80, 2024-09-03:
For small weights it seems to work with autocalibration.
But if I later put large weights on the scale, the error is very large, several kilograms...
For example, if I calibrate with a weight of 2000 grams, after calibration I measure 2070 - 2073 grams.
And then if I get on the scale with 75200 grams, the scale displays 79680 - 79683 and it annoys me because I'm not even like a bear :)
Calibrated scale value can only be an integer?
I tried to enter 11.4 or 11.4, but the saved value remains fixed at 11...
Can someone give me an idea how to properly calibrate the scale for a weight up to a maximum of 100KG?

Thank you