arendst/Tasmota #4843: Correctness of Hardware Preparation wiki description of sonoff power supplies

lbdroid, 2019-01-06:
I've corrected an error in the wiki describing dangers related to the power supply of itead sonoff AC power devices. The wiki was incorrectly implying that there was AC current on the digital interface header, and that use of this header while the sonoff is connected to AC power would destroy the sonoff and potentially the device being connected to it. The wiki even went so far as to provide a picture of a damaged sonoff, presumably due to interfacing with the low voltage components while connected to AC -- however, due to the circuit design, this is actually impossible. The most likely explanation for that picture is that someone was making the connection with something similar to male dupont connectors and one slipped out of place and shorted the board out.

My own use of sonoff devices necessitates a perpetual connection with the data header, and that warning threw me off for a while until I studied the device schematic and realized that it was complete nonsense.

Sonoff devices have isolated power supplies. From a hardware design point of view, it is perfectly safe to interface with the low voltage components while it is powered by AC, just as long as the sonoff's low voltage ground is connected first (or prior to being powered) to the devices being connected with it, rather than data pins.

Now I've never made any claims about the safety of playing with exposed AC circuits if you aren't ABSOLUTELY competent in manipulating them, so I have no objections to warnings about the personal risks involved, but I find it offensive when somebody would revert the changes back to the incorrect version immediately after it is made correct. And I understand that it is a wiki, which makes policing it challenging.

arendst, 2019-01-06:
And did you look at the Sonoff Pow. This device definitely has NO ISOLATED POWER SUPPLY.

I suggest you buy a Sonoff Pow. Connect it to AC and connect your header to the appropriate connector. Preferably a laptop. If nothing happens pls turn the pow power connector 180 degrees and watch your laptop going up in flames.

Pls report back if you made it.

andrethomas2, 2019-01-06:
Closing this issue as it has been answered and we have not seen cases of users responding to issues from the afterlife...

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-01-06 by andrethomas2

ascillato, 2019-01-06:
I revert your changes and add more warnings because we already had issues of people complaining about electroshocks.

There are a bunch of newbies that know nothing about AC. So, if you (lbdroid) knows what you are doing with the hardware, I congratulate you. But for most of the people "tinkering" with sonoff devices, a warning for danger MUST BE MADE.

All the sonoffs and other brands of devices with power measurements HAVE THE MAINS AC DIRECTLY CONNECTED TO THE ESP8266, so any sensor or switch you connect WILL BE ALSO CONNECTED TO MAINS AC.

I know, a sonoff basic have the power regulator isolated from mains, but a pow doesn't.

So, please, don't get mad or complain about this warnings we made. We don't want any people to get electrocuted because they are tinkering without knowledge.

SAFETY FIRST.

Please, do not revert the warnings of the wiki or I will put this case to github to ban your account for unsafety and dangerous comments.

lbdroid, 2019-01-06:
@arendst : Really....
Take a peek on PAGE 1: https://www.itead.cc/wiki/images/5/52/Sonoff_POW_Schematic.pdf

Component EE10 in grid C3. Left side is high voltage. Right side is low voltage.

lbdroid, 2019-01-06:
@ascillato : I have no objection at all about warning noobs about playing around with AC, but lying to them about why is NOT an appropriate method of getting that concept across to them.

ascillato, 2019-01-06:
There is no LYING about the REAL DANGER or connecting sensors to a POW.

Please, **_last warning until I put this case to GITHUB._**

Please, DO NOT CHANGE ANYMORE THE WIKI and please TEST the POW for this.

arendst, 2019-01-06:
@lbdroid while talking about this schematic I suggest you start following the L marked signal...

You might find something like R10 (0.001R) going from L to L0. Then looking further to the picture you might also come across R25 (0R) connecting the same L0 to logic Gnd, the same you refer to as safe.

Again, connect up your gear and enjoy the free fireworks.

Einstein67, 2019-01-06:
Blitzwolf SHP2

[image]

Has the same problem as the Sonoff-POW (sorry no picture made). 236VAC  at the GPIOs and GND, 233VAC at Vcc ---> that ist dangerous!!