arendst/Tasmota #4844: Shelly 1 Tasmota - Need help to reset wifi

bspiegel33, 2019-01-06:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Development/Compiler/Upload tools used :
- [ ] Hardware used : 
- [ ] If a pre-compiled release or development binary was used, which one? :
- [ ] You have tried latest release or development binaries? :
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2019-01-06:
Hi,

please, for support use the Tasmota Support Chat.

Github issues are for bug reporting. Thanks.

bspiegel33, 2019-01-06:
I have a Shelly 1 flashed with Tasmota 6.4.1.  All things working.  I went to change the host name and it must have also changed the wifi settings because it is no longer connecting to my wifi.  I'd like to reset wifi to Access Point mode or find another way to get access.  

The Shelly is connected to a switch which performs as a toggle.  I've tried to simulate button presses with the switch to enter AP mode ( 4 quick on/off) but no luck.  My starting position for the switch was OFF.  

The second AP setting was never setup.  If there is a default on that, I can setup a router so Shelly can find it.  All I need is what the defaults are.

I am hoping to avoid dismantling the hardware to reset this.

Much appreciate any help.

closed 2019-01-06 by bspiegel33