arendst/Tasmota #4846: [How to?] HTTP page with ON/OFF Button

Lusitanv, 2019-01-06:
Hello,

I have now 3 sonoff basic flashed with TASMOTA.

I would like to write an HTML page and store it in a server to ease the access with simple ON and OFF buttons for all the sonoffs.

I can access them via WAN and LAN.

Could you please point me in that direction? What would be my HTML code?

This would avoid the need to access to the admin page, for each and then use the page buttons...

Cheers and thanks in advance!

curzon01, 2019-01-06:
Try (TasmoAdmin)[https://github.com/reloxx13/TasmoAdmin]

andrethomas2, 2019-01-06:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-01-06 by andrethomas2

Lusitanv, 2019-01-08:
Why was this closed?

I understand the first "solution" but I simply would like to know the html code to add a button to a page to send the following commands:

http://sonoff/cm?cmnd=Power%20TOGGLE
http://sonoff/cm?cmnd=Power%20On
http://sonoff/cm?cmnd=Power%20off
http://sonoff/cm?user=admin&password=joker&cmnd=Power%20Toggle

This would allow me to have a simple webpage with ON or OFF and thats it

curzon01, 2019-01-08:
'cause this is a issue list, not a forum - your question has nothing to do with Tasmota; it's a question related to html.
Please use the Chat linked above for that

Lusitanv, 2019-01-08:
Makes sense.
Will do.

cgd1, 2020-07-28:
@Lusitanv did you manage to create the html page ?? :)

joba-1, 2020-07-28:
maybe this is a start?

`<html>
        <head>
        </head>
        <body>
                <form method="post" action="http://192.168.1.120/cm">
                        <input type="submit" name="cmnd" value="Power On">
                        <input type="submit" name="cmnd" value="Power Off">
                </form>
        </body>
</html>
`

kugelkopf123, 2020-07-28:
There is a possibility with the scripter. https://tasmota.github.io/docs/Scripting-Language/

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

cgd1, 2020-07-28:
thank you bot kindly for replies :+1: 

@joba-1 I like your way however I found it a little constricting because it redirects to response

I made a script for the html to do the call in the end, I guess its a more mature way to do it
```
<html>
        <head>
        </head>        
        <body>
        <input id="contact-submit" type="button" class="btn btn-transparent" value="Submit" onclick="toggle()" />
        </body>
<script>
function toggle() {
    var http = new XMLHttpRequest();
    var url = 'http://192.168.1.187/cm?cmnd=Power%20TOGGLE';
    http.open('POST', url, true);
    http.onreadystatechange = function() {//Call a function when the state changes.
        if(http.readyState == 4 && http.status == 200) {
            //alert(http.responseText);
        }
    }
    //content you could return as alert?
    //var content = {"POWER" : "On"};
    //http.send(JSON.stringify(content));
    http.send();
}
<script>
</html>
```

JardaG, 2020-10-17:
More advanced HTML version. :-)
```
<html>
  <head>
    <style type="text/css">
      button {height: 80px; width: 300px;}
      button.on {background-color: green;}
      button.off {background-color: red;}
      button.toggle {background-color: orange;}
    </style>
  </head>
  <body> 
    <a href="http://192.168.0.1/">Tasmota GUI</a>
    <div id="192.168.0.1">
      <button class="On" onclick="tasmota(this)">On
      </button>
      <button class="Off" onclick="tasmota(this)">Off
      </button>
      <button class="Toggle" onclick="tasmota(this)">Toggle
      </button>
    </div>
  </body>
<script>
// All Tasmota commands: https://tasmota.github.io/docs/Commands/
// Open: http://192.168.0.1/cs?
// CORS " = disable CORS (Cross Origin Resource Sharing) (default)
//      * = enable CORS for all locations
function tasmota(element) {
    url = 'http://' + element.parentNode.id + '/cm?cmnd=Power%20' + element.className
    //alert(url) // debug
    var http = new XMLHttpRequest();
    http.open('post', url, true);
    http.onload = function() { alert (http.responseText); } // success case
    http.onerror = function() { alert (http.responseText); } // failure case
    http.send();
}
</script>
</html>
```